ev3dev/ev3dev #1622: EV3 with Multiplexer

Alessandro-sys, 2024-03-14:
Ok, tell me if this makes any sense
I need to use 2 EV3 bricks with ev3dev and the bluetooth communication is unusable. 
I want to use an ev3 Sensor Multiplexer to make them comunicate with i2c. It has 4 ports. I connect ports 1, 2, 3 of the first brick to sensors. The 4 port is connected to the in port of the multiplexer. The first port of the multiplexer is connected to the first port of the second brick. To the 3 e 4 port of the multiplexer I have 2 more sensors and on port 2, 3, 4 of the second brick I have 3 more sensors. Maybe an image would be more easy to comprehend: 
[image]
(sorry for my bad drawing skills)
Would this be possible? if yes, how?

dlech, 2024-03-14:
> and the bluetooth communication is unusable.

why? how about USB as an alternative?

> Would this be possible? if yes, how?

Not without some serious hacking of the PRU firmware and kernel drivers. The problem is that I2C has a host side and a peripheral side that connect to each other. Both the EV3 and C1 on the multiplexer implement the host side so they can't be connected together and actually communicate. The EV3 would have to be changed to be a peripheral, but this hasn't been implemented in the software.

Alessandro-sys, 2024-03-14:
> how about USB as an alternative?

is it possibile? still ev3dev and micropython I mean

Alessandro-sys, 2024-03-14:
I know that with the ev3 software it's possibile, with daisy chaining, but how about ev3dev?

dlech, 2024-03-14:
In ev3dev, you can have a network connection over USB between two bricks just like you can from the EV3 to your computer.

If you are using Pybricks MicroPython for EV3, there are Bluetooth mailbox APIs for communicating via Bluetooth. It could probably be made to work with vanilla MicroPython as well.

Alessandro-sys, 2024-03-14:
is there any link to show me how?

dlech, 2024-03-14:
I don't know of any good links. But there are a number of issues here about daisy chaining. I think the trick is to enable the USB gadget feature on one of the EV3's. If each EV3 has a unique hostname, you should be able to find the other EV3 and create a network socket using the hostname.

Alessandro-sys, 2024-03-15:
ok, I have no idea how to do this, nice :)