ev3dev/ev3dev #1624: ev3 refuses to connect to the internet

FunnyIvri, 2024-03-25:
**What are you trying to do?**
connect my ev3 to the internet
I am trying to ...?
I'm using Ubuntu 22.04.4 LTS i connect the ev3 to my laptop i go to advanced network configuration and change the name of the connection to ev3deb and change method to "Shared to other computers" then i go on the ev3 "Wireless and Networks" -> "All Network Connections" -> "Wired" and click connect
**What did you expect to happen**
the text changes from disconnected to online
The program should have...
connected the ev3 to the internet through my laptop
**What actually happened**
the ev3 immediately goes too "connected" and `ssh robot@ev3dev.local` doesn't work

**Desktop**
 - OS and version: Ubuntu 22.04.4 LTS GNOME 42.9
 - Connection Type: USB
 - Software: Linux terminal, advanced network configuration

**Robot**
 - Device: EV3
 - Versions: ev3dev-stretch-ev3-generic-2020-04-10

**Background Information**
everything was working fine the ev3 was online and it worked then it just stopped.

dlech, 2024-03-25:
What is the error message?

Does the EV3 show an IP address? If yes, what is it?

Does it work if you replace `ev3dev.local` with the IP address?

FunnyIvri, 2024-03-25:
the ev3 ip address is: 10.42.0.3:
(base) funny@usb-ubntu:~$ ssh robot@10.42.0.3
(robot@10.42.0.3) Password: 
Linux ev3dev 4.14.117-ev3dev-2.3.5-ev3 #1 PREEMPT Sat Mar 7 12:54:39 CST 2020 armv5tejl
             _____     _
   _____   _|___ /  __| | _____   __
  / _ \ \ / / |_ \ / _` |/ _ \ \ / /
 |  __/\ V / ___) | (_| |  __/\ V /
  \___| \_/ |____/ \__,_|\___| \_/

Debian stretch on LEGO MINDSTORMS EV3!
Last login: Sun Mar 24 22:48:04 2024 from 10.42.0.1
robot@ev3dev:~$ ping 8.8.8.8 -c 10
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=117 time=9.75 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=117 time=10.8 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=117 time=8.78 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=117 time=8.40 ms
64 bytes from 8.8.8.8: icmp_seq=5 ttl=117 time=6.27 ms
64 bytes from 8.8.8.8: icmp_seq=6 ttl=117 time=7.67 ms
64 bytes from 8.8.8.8: icmp_seq=7 ttl=117 time=9.79 ms
64 bytes from 8.8.8.8: icmp_seq=8 ttl=117 time=8.49 ms
64 bytes from 8.8.8.8: icmp_seq=9 ttl=117 time=8.76 ms
64 bytes from 8.8.8.8: icmp_seq=10 ttl=117 time=8.44 ms

--- 8.8.8.8 ping statistics ---
10 packets transmitted, 10 received, 0% packet loss, time 9014ms
rtt min/avg/max/mdev = 6.273/8.720/10.807/1.180 ms
the ev3 still says "State: Connected"

FunnyIvri, 2024-03-25:
* note that when run `ping google.com` it says:
ping: google.com: Name or service not known

dlech, 2024-03-25:
Hmm... it sounds like it is connected to the Internet, but domain name resolution isn't working. Maybe you could try adding a different DNS server?

FunnyIvri, 2024-03-25:
i tried but when i did the ev3 just froze
ill wait a bit but its just frozen

FunnyIvri, 2024-03-25:
i think my only option is to take out the battery

dlech, 2024-03-25:
You can press and hold the center and back buttons to reset.

FunnyIvri, 2024-03-25:
welp too late....
so do you know why it freezes when i try to add a new dns?

dlech, 2024-03-25:
No idea. I haven't seen that before.

dlech, 2024-03-25:
Unless maybe there is an issue with the SD card (e.g. > 32GB or defective).

FunnyIvri, 2024-03-25:
i have a 32gb a1 sdhc sandisk sdcard

FunnyIvri, 2024-03-25:
i could try to reflash it

dlech, 2024-03-25:
It was just a wild guess. Not sure if it is worth re-flashing. But, you never know...

FunnyIvri, 2024-03-25:
i did it didnt help... tried everything again still cant ping google.com when i try to add a new dns it freezes

FunnyIvri, 2024-03-25:
very weird as everything work yesterday i didnt change anything