ev3dev/ev3dev #1626: How do you use pybricks from the LEGO-approved version of ev3dev?

rayluo, 2024-04-01:
**Question**

First of all, a big THANK YOU to @dlech for providing this project! I'm new to ev3dev, and it already looks like lots of fun ahead of me. :-)

By the way, I read [your another answer from another issue](https://github.com/ev3dev/ev3dev/issues/1587#issuecomment-1699686185): 
> **To fix apt in general**, `/etc/apt/sources.list` needs to be changed to match https://github.com/ev3dev/docker-library/blob/master/ev3dev-stretch/layers/debian/etc/apt/sources.list

The suggestion above was helpful! It took me a while reading some outdated "sudo apt update" issues, before stumbling upon this gem. Where else is this piece of important information available? I would suggest converting this into a Q&A in Github Discussion.

Now, my question is that, I installed the "LEGO-approved" version of ev3dev, because the image file was one month newer than the [non-LEGO ev3dev](https://www.ev3dev.org/downloads/). Isn't the LEGO-approved ev3dev supposed to come with Micro-Python? I followed some of [LEGO documents](https://pybricks.com/ev3-micropython/startrun.html#creating-a-new-project) to tentatively do `python3 -c "import pybricks"`, but that Python package was unavailable.

**Background Information**

I am trying to explore whether I shall go with the `ev3dev-lang-python` or the `pybricks-micropython`, because I am new to ev3dev and want to choose a better path forward.

**Screenshots**

N/A

**Desktop (please complete the following information):**
 - OS and version: Linux
 - Connection Type: Wi-Fi, and the brick has Internet via Wi-Fi
 - Software: Just good old SSH so far

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3-micropython-v2.0.0-sd-card-image
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534C0D06
BOARD_INFO_TYPE=main
```

dlech, 2024-04-01:
>  Where else is this piece of important information available?

Just buried in some other issues. We should probably update the web page with this. (Pull requests welcome :smile:)

> Isn't the LEGO-approved ev3dev supposed to come with Micro-Python?

Yes, it is nearly identical to the regular ev3dev image. It comes with both "vanilla" MicroPython and Pybricks MicroPython (I'm guessing you ware wanting the latter).

> do `python3 -c "import pybricks"`, but that Python package was unavailable.

Python (and therefore `python3` is not MicroPython. Try  `brickrun -r pybricks-micropython -c "import pybricks"` 

Or even better (if you know what `>>>` is with regard to Python :wink:)... `pybricks-repl`.

> shall go with the `ev3dev-lang-python` or the `pybricks-micropython`

It depends on if you want features or performance. `ev3dev-lang-python` uses "real" Python, so can do many more things, but runs rather slow on the EV3. `pybricks-micropython` is optimized for performance (except when using fonts to render text to the display) and has really nice motor control, but lacks other general features because it is MicroPython. You can even have the best of both worlds if you want by creating two scripts, on with real Python and one with Pybricks MicroPython that communicate with each other.

rayluo, 2024-04-02:
Thanks for your quick response, @dlech !

> We should probably update the web page with this. (Pull requests welcome 😄)

I was not aware that the web pages are also in an open github repo. I am more than willing to create a PR to add those description into [the Download page](https://github.com/ev3dev/ev3dev.github.io/blob/master/downloads.md). Is that the best approach here? I thought there could be enough update for a new release after 4 years, so that those kind of documentation-patch wouldn't be necessary.

> Python (and therefore python3 is not MicroPython. Try brickrun -r pybricks-micropython -c "import pybricks"

Thanks for your hint! I came from the standard Python world, and was not aware that `micropython` (and `pybricks-micropython`, for that matter) is a standalone interpreter rather than a package. Now, it looks like package `ev3dev2` and `pybricks` cannot coexist within same Python program. They have to be two different programs, potentially talking to each other via inter-process communications. OK, I can live with that.

dlech, 2024-04-02:
> Is that the best approach here?

Yes, and also the getting started page (one could just link to the other so we don't have to repeat everything).

> I thought there could be enough update for a new release after 4 years, so that those kind of documentation-patch wouldn't be necessary.

If I can find the time and energy. It's been been so long the release process is a bit broken (and perhaps forgotten).

rayluo, 2024-04-09:
Thanks for your code review, @dlech . Now that the Getting Started page has been changed, let's focus back on the "[there is no fix for outdated image](https://github.com/ev3dev/ev3dev.github.io/pull/442#discussion_r1555895539)".

* If you can remember and document how a new release could be built, I am willing to do the work for you. Please help me to help you. :-)
* Personally, I focus more on the Python at this moment. So, a newer version of Python (3.8+) running on an older Debian - if at all possible - would also be better than status quo. Would that be slightly easier than upgrading the entire image, David?

rayluo, 2024-04-09:
> There have been some folks playing around with updating to a newer Debian version in #1554 so maybe you could join them?

Thanks for the hint. I'll take a look.

Meanwhile, the current github issue (and perhaps many others) shall be closed, so that people shall not get a misperception that this repo has hundreds of open issues. Let me know if you'd like to:

1. enable [Github Discussions](https://docs.github.com/en/discussions/collaborating-with-your-community-using-discussions/about-discussions) for this repo so that we can convert this issue into a Q&A with a chosen answer; or
2. Summarize the how-to from [here](https://github.com/ev3dev/ev3dev/issues/1626#issuecomment-2030703114) and [there](https://github.com/ev3dev/ev3dev/issues/1626#issuecomment-2030901663) and put them into yet another PR for [the programming languages page](https://www.ev3dev.org/docs/programming-languages/).

dlech, 2024-04-10:
This repo was setup long before github discussions were a thing, so most of the issues here in the last few years are just questions. So I'm not sure there is a lot to be gained by switching now given the relatively low traffic on the project.

Happy to take any more improvements to the web pages though to keep those up to date.

Growflavor, 2024-04-18:
> > There have been some folks playing around with updating to a newer Debian version in #1554 so maybe you could join them?
> 
> Thanks for the hint. I'll take a look.
> 

@rayluo 

More tinkerer's are greatly welcome ;-)  

--The packages built for Stretch seem to run reasonably well with the Buster image & at least the EV3dev2 package seems to run OK with the 'Bullseye' on the 4.19 kernel old beta which has Python 3.9.2...though my son & I have only done basic 'line follower' & RPyC type things so far with that image...I provided a link to that 'tinker' image in a post at the 1554 thread @dlech mentioned.

--and/or, you can fairly easily use docker to pull & update in a docker container @dlech 's [Buster or Bullseye Beta](https://github.com/ev3dev/docker-library) & use [Brickstrap](https://github.com/ev3dev/brickstrap) to turn them into flash-able SDcard images just like I did.

--EV3 sell like hotcakes on US ebay...and for introducing an ~age 12 person to 'real' tools of data science, robotics, ML, and just plain fun of controlling an EV3 via VScode & RPyC Jupyter notebook cells, the EV3dev project has made the EV3 a remarkable & fantastic device.  I very much agree RE: "a big THANK YOU to @dlech"!

--Also, I suggest checking out the online "[Generic Educational Autonomous Robotics Simulator](https://github.com/QuirkyCort/gears)" by @QuirkyCort -- https://quirkycort.github.io/gears/public/
 
--The armel Debian Kernel aspect is well beyond me...but I noticed that @StepanTheFlowey has been doing something with[ a 4.19 kernel fork](https://github.com/StepanTheFlowey/ev3-kernel).

closed 2024-05-05 by rayluo