ev3dev/ev3dev #1628: Gyiro sensor isn't detected

JSDaleman, 2024-04-16:
**What are you trying to do?**

I am trying to get information with the gyro sensor to use a script with the example 

**What did you expect to happen**

The program should have move the robot

**What actually happened**

The program did have a error because the sensor is not dectect in any port
I am working with 
```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          2.1
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165341C898
BOARD_INFO_TYPE=main
```

**Source Code**

Here is the program I am trying to run.

```
from ev3dev2.motor import OUTPUT_A, OUTPUT_B, MoveTank, SpeedPercent, follow_for_ms
from ev3dev2.sensor.lego import GyroSensor

# Instantiate the MoveTank object
tank = MoveTank(OUTPUT_A, OUTPUT_B)

# Initialize the tank's gyro sensor
tank.gyro = GyroSensor()

# Calibrate the gyro to eliminate drift, and to initialize the current angle as 0
tank.gyro.calibrate()

try:

    # Follow the target_angle for 4500ms
    tank.follow_gyro_angle(
        kp=11.3, ki=0.05, kd=3.2,
        speed=SpeedPercent(30),
        target_angle=0,
        follow_for=follow_for_ms,
        ms=4500
    )
except FollowGyroAngleErrorTooFast:
    tank.stop()
    raise
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS and version: [e.g. Window 10 1809, macOS 10.14]
 - Connection Type: [e.g. USB, Bluetooth, Wi-Fi]
 - Software: [e.g. Visual Studio Code, PuTTY]

I am working whit visual Studio code and a wifi conection

**Robot (please complete the following information):**
 - Device: [e.g. EV3, BrickPi3]
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          2.1
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165341C898
BOARD_INFO_TYPE=main

```

**Background Information**

I am trying to ... because I want to build a robot that ...

Growflavor, 2024-04-18:
Check if you can see the gyro via the Brickman menu...
If not, then usually it is a bad wire connection...I often need to re-plug wires to get sensors to register...

If you can see it in Brickman, but still have trouble in your program, you can also try explicitly specify which input port to look at for the gyro...

JSDaleman, 2024-04-18:
how can i check the Gyro sensor via the brickman menu ?

dlech, 2024-04-18:
By looking in the *Device Browser*.

JSDaleman, 2024-04-18:
> By looking in the *Device Browser*.

The Gyro sensor isn't detected, i change the sensor, others sensors with the same cable are detected and with other cable Gyro still has the problem

juliovendramini, 2024-04-18:
Try to use with original Lego Firmware. And check if there is working

JSDaleman, 2024-04-18:
On the original firmware is detected and work normal

Bricklou, 2024-05-15:
I can confirm the issue : the gyro sensor is detected in the original firmware. But in ev3dev firmware, it is displayed as uart in the ports section and not showing in the sensors section.

Bricklou, 2024-05-29:
I began to try looking at the issue, here is the `dmesg | tail` output:
```
[  234.476041] lego-port port1: Removed device 'ev3-ports:in2:ev3-uart-host'
[  236.806583] lego-port port1: Added new device 'ev3-ports:in2:ev3-uart-host'
```

It seems the kernel doesn't detect the gyro sensor as it should.

Growflavor, 2024-06-08:
1) Neither of you see this in Brickman?:
[image]

2) when you run systmctl do you see (I have a gyro at 'in1'):

` sys-devices-platform-ev3\x2dports-ev3\x2dports:in1-lego\x2dport-port0-ev3\x2dports:in1:ev3\x2duart\x2dhost.device loaded active plugged   /sys/devices/platform/ev3-ports/ev3-ports:in1/lego-port/port0/ev3-ports:in1:ev3-uart-host

  sys-subsystem-lego-devices-ev3_uart_host_ev3\x2dports:in1.device                                                  loaded active plugged   /sys/subsystem/lego/devices/ev3_uart_host_ev3-ports:in1    `

3) Do you see this firmware version installed?
robot@ev3dev:~$ apt-cache policy firmware-ev3
firmware-ev3:
  Installed: 6
  Candidate: 6
  Version table:
 *** 6 990
        990 http://archive.ev3dev.org/debian stretch/main armel Packages
        100 /var/lib/dpkg/status
robot@ev3dev:~$ 

4) This modified [example motor script](https://ev3dev-lang.readthedocs.io/projects/python-ev3dev/en/stable/motors.html) shown in a prior post executes fine for me with a the [standard EV3 drive base with gyro](https://assets.education.lego.com/v3/assets/blt293eea581807678a/bltc961107e6587a579/5f8801dd3b19d21d4a6085bf/ev3-gyro-sensor-driving-base.pdf?locale=en-us):

<details>
  <summary>Tank Gyro Test</summary>

Script below runs OK on this apt updated EV3dev version:
========== ev3dev-sysinfo ==========
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          2.1
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016533D57B3
BOARD_INFO_TYPE=main
  
  ```javascript
#!/usr/bin/env python3

# Import the necessary libraries
import time
import math
#from ev3dev2.motor import *
from ev3dev2.sound import Sound
from ev3dev2.button import Button
from ev3dev2.sensor import *
#from ev3dev2.sensor.lego import *
from ev3dev2.motor import OUTPUT_A, OUTPUT_B, LargeMotor, MoveTank, SpeedPercent, follow_for_ms, FollowGyroAngleErrorTooFast
from ev3dev2.sensor.lego import GyroSensor

# Create the sensors and motors objects
motorA = LargeMotor(OUTPUT_A)
motorB = LargeMotor(OUTPUT_B)
left_motor = motorA
right_motor = motorB
#tank_drive = MoveTank(OUTPUT_A, OUTPUT_B)
#steering_drive = MoveSteering(OUTPUT_A, OUTPUT_B)

spkr = Sound()
btn = Button()
#radio = Radio()

#color_sensor_in1 = ColorSensor(INPUT_1)
#ultrasonic_sensor_in2 = UltrasonicSensor(INPUT_2)
#gyro_sensor_in3 = GyroSensor(INPUT_3)
#gps_sensor_in4 = GPSSensor(INPUT_4)
#pen_in5 = Pen(INPUT_5)

#motorC = LargeMotor(OUTPUT_C) # Magnet

# Here is where your code starts

# Instantiate the MoveTank object
tank = MoveTank(OUTPUT_A, OUTPUT_B)

# Initialize the tank's gyro sensor
tank.gyro = GyroSensor()

# Calibrate the gyro to eliminate drift, and to initialize the current angle as 0
tank.gyro.calibrate()

try:
    # Follow the target_angle for 4500ms
    tank.follow_gyro_angle(
        kp=11.3, ki=0.05, kd=3.2,
        speed=SpeedPercent(30),
        target_angle=0,
        follow_for=follow_for_ms,
        ms=4500
    )
except FollowGyroAngleErrorTooFast:
    tank.stop()
    raise
else:
      tank.follow_gyro_angle(
        kp=11.3, ki=0.05, kd=3.2,
        speed=SpeedPercent(30),
        target_angle=0,
        follow_for=follow_for_ms,
        ms=4500
    )

  ```
</details>

Bricklou, 2024-06-09:
I reproduce everything you did in steps 1,2 and 3. I get the exact same results for step 2 and 3. But i don't have the gyro in the sensors list like on step 1.