ev3dev/ev3dev #1629: How can I add a driver I wrote for an I2C sensor to the kernel?

BeatThisStone, 2024-05-12:
- **ev3dev version:** 4.14.114-ev3dev-2.3.5-rpi2
- **ev3dev-lang-python version:** micropython-ev3dev2 2.1.0, python3-ev3dev 1.2.0, python3-ev3dev2 2.1.0

Following some guides I discovered you can add a driver editing `nxt_i2c_sensor.h` and `nxt_i2c_sensor_defs.c` in the [lego-linux-drivers](https://github.com/ev3dev/lego-linux-drivers/tree/c768f33667c4220fbb3163143de61ebcd3e1e424/sensors) Sensor folder.

After doing so, what do I do with the resulting files?

dlech, 2024-05-12:
What sensor do you want to write a driver for? 

Unless this is a MINDSTORMS compatible sensor that has an ID readable at a specific register, there probably isn't much point in trying to add it to the kernel drivers. Instead, it is easier to use the userspace I2C interface to communicate with the chip, e.g. with the Python SMBus library.

BeatThisStone, 2024-05-12:
The sensor is this IR Seeker https://irseeker.buildingblockrobotics.com/

dlech, 2024-05-12:
It looks like they already have a way to use it with ev3dev using Pybricks MicroPython.

https://irseeker.buildingblockrobotics.com/guides/ev3#ev3-python

I didn't see any documentation on the register layouts, but it sounds like this sensor doesn't follow the usual layout.

BeatThisStone, 2024-05-12:
So no can do?

BeatThisStone, 2024-05-13:
Thanks a lot.

closed 2024-05-13 by BeatThisStone