ev3dev/ev3dev #1630: bluetooth issues

FunnyIvri, 2024-05-21:
**What are you trying to do?**
control my robot with a steering wheel

**What did you expect to happen**
my program is a server client setup i have a server on my laptop that is connected to the steering wheel
and a client running on the ev3

**What actually happened**
it refused to connect

**Source Code**

- server.py
from socket import socket, AF_BLUETOOTH, SOCK_STREAM, BTPROTO_RFCOMM
from keyboard import is_pressed
import hid
from ManageData import EncryptData
from wheelApi import wheelFinder, getInput
s = socket(AF_BLUETOOTH, SOCK_STREAM, BTPROTO_RFCOMM)
try:
    device = wheelFinder()
    gamepad = hid.device()
    gamepad.open(device['vendor_id'], device['product_id'])
    gamepad.set_nonblocking(True)
    print('server starting up') 
    hostMACAddress = '00:17:e9:b3:78:5d'
    port = 1
    backlog = 1
    size = 1024
    
    print(s)
    s.bind((hostMACAddress, port))
    s.listen(backlog)
    print('server ready')
    
    clientSocket, clientAddress = s.accept()
    print(f'new ev3[{clientAddress}]')
    i = 0
    
    while True:
        print(f'loop: {i}')
        i += 1
        if is_pressed('q'):
            clientSocket.send(EncryptData(f'q 0'))
            break
        
        wheel, pedals = getInput(gamepad)
        data = f'{wheel} {pedals}'
        print(data)
        clientSocket.send(EncryptData(data))
        
        status = bool(clientSocket.recv(1024))
        print(status)
        
        if not status:
            clientSocket.send(EncryptData("q"))
            print(f'ev3[{clientAddress}] has left!')
            break

except Exception as error:
    print(f'error!\n {error}')
finally:
    s.close()

- client.py
#!/usr/bin/env python3
import socket
from ev3dev2.motor import MoveSteering, OUTPUT_B, OUTPUT_C
from ManageData import DecryptData
try:
    wheelSensitivity = 90
    pedalsSensitivity = 50
    print('starting ev3dev')
    motor_pair = MoveSteering(OUTPUT_B, OUTPUT_C)
    print('ev3dev ready')
    print('starting server')
    
    serverMACAddress = '00:17:e9:b3:78:5d'
    client = socket.socket(socket.AF_BLUETOOTH, socket.SOCK_STREAM, socket.BTPROTO_RFCOMM)
    client.connect((serverMACAddress,9999))
    print('client ready')
    while True:
        data = DecryptData(client.recv(1024))
        if data[0] == "q":
            break
        else:
            steering = data[0] * wheelSensitivity
            speed = data[1] * pedalsSensitivity
            motor_pair.on(steering,speed)
            client.send(str(True).encode('utf-8'))
            
except Exception as e:
    open('log.txt','w').write(str(e))
    print(e)
    
finally:
    client.close()

**Desktop**
 - OS and version: Windows 11
 - Connection Type: Bluetooth
 - Software: vs code

**Robot**
 - Device: EV3
 - Versions: 2020-04-10

dlech, 2024-05-21:
What is the actual error you are getting?

FunnyIvri, 2024-05-21:
error!
 [WinError 10013] An attempt was made to access a socket in a way forbidden by its access permissions

dlech, 2024-05-21:
Windows uses RFCOMM channel 1 automatically to create a Bluetooth serial port. You should be able to change `port` to a higher number to get a free RFCOMM channel.

FunnyIvri, 2024-05-21:
[WinError 10049] The requested address is not valid in its context

FunnyIvri, 2024-05-21:
[image]

dlech, 2024-05-21:
Which line throws the error?

FunnyIvri, 2024-05-22:
Found the problem, there where problems with the way i got input and also i needed to get the adapters WW2 address not the ev3s for anyone wanting to do some robot driving ill upload the code to github today

closed 2024-05-22 by FunnyIvri

FunnyIvri, 2024-05-22:
done, cleaned up the code, if anyone wants heres the source code: https://github.com/FunnyIvri/EV3SteeringWheel