ev3dev/ev3dev #1631: Error using Ev3 Micropython.

Robocup-Programmer-QASMT, 2024-06-01:
- **ev3dev version:** PASTE THE OUTPUT OF `uname -r` HERE
- **ev3dev-lang-python version:** INSERT ALL VERSIONS GIVEN BY `dpkg-query -l {python3,micropython}-ev3dev*` HERE

My version: the one downloaded from the lego Ev3 Website for education. 2.0

Attached is the code I am using. It is For the RoboCup junior rescue line competition. When running the code, the console gives a message saying there is an error using commands like  .stop and .brake. The error is happening every second time. The message says something about the directory: Note. Couldn't access this website at school when preparing for comp and was unable to get the error msg. 
File copied as unable to add.

```python
#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch, DataLog
from pybricks.robotics import DriveBase
from pybricks.media.ev3dev import SoundFile, ImageFile

# This program requires LEGO EV3 MicroPython v2.0 or higher.
# Click "Open user guide" on the EV3 extension tab for more information.

# Create your objects here.
ev3 = EV3Brick()
left_motor = Motor(Port.A)
right = Motor(Port.D)
robot = DriveBase(left_motor,right,wheel_diameter=55.5, axle_track=104)
color_left = ColorSensor(Port.S4)
color_right = ColorSensor(Port.S1)
#searchrescue = UltrasonicSensor(Port.S3)
#retrieveright = Motor(Port.C)
#retrieveleft = Motor(Port.B)
# Write your program here.
#In this program, the robot sees all color as shades of gray. If both color sensors detect white, the robot will move forward.
#If one sensor detects 
while True:
   #if color left detects white
   if color_left.reflection() > 50:
       #if both sensors detect white, drive
       if color_right.reflection() > 50:
           robot.drive(100,0)
        #else if color right detects black, turn right
       elif color_right.reflection() < 50:
            right.stop()
            left_motor.run(100)
    #if color left detects black and;
   elif color_left.reflection() < 50:
        #color right detects white, turn left
        if color_right.reflection() > 50:
            robot.stop()
            robot.drive(100,100)
        #else if both sensor detect black, look for either can (both sensors green) or the hint (one sensor green)
        elif color_right.reflection() < 50:
            robot.stop()
            if color_left.color() == Color.GREEN and color_right.color() == Color.GREEN:
                robot.brake()
                robot.straight(60)
                robot.brake()
                while capsulefound == 0:
                    if searchrescue.distance() < 100:
                        robot.straight(ultrasonic.distance-10)
                        #retrieve.run_target(40,90)
                        #find and follow green line to exit
                        #use color instead of reflected light intensity
                        robot.drive (40,0)
                        while outofspill == 0:
                                  #robot will now run line following code with slight modifications to exit chem spill                               
                            if color_left.reflection() > 50 or color_right.reflection() > 50:
                                     robot.straight(20)
                                     robot.turn(-90)
                            elif color_left.color() == Color.GREEN and color_right.color() == Color.WHITE:
                                        robot.drive(40,0)
                            elif color_left.color() == Color.GREEN and color_right.color() == Color.GREEN:
                                        right.brake()
                                        left_motor.run(40)
                            elif color_left.color() == Color.WHITE:
                                        left_motor.brake()
                                        right.run(40)
                            elif color_left.color() == Color.GREEN and color_right.color() == Color.BLACK:
                                        robot.brake()
                                        robot.turn(90)
                                        robot.straight(40)
                                        robot.brake()
                                        robot.turn(90)
                                        retrieveright.run_target(40,-90)
                                        retrieveleft.run_target(40,-90)
                                        robot.turn(-90)
                                        outofspill = 1
                                        capsulefound = 1
                    else:
                        robot.turn(10)
            if color_left.color() == Color.GREEN:
                robot.brake()
                robot.straight(10)
                robot.turn(-90)
            elif color_right.color() == Color.GREEN:
                robot.brake()
                robot.straight(10)
                robot.turn(90)
            else:
                robot.straight(40)
  ```

BertLindeman, 2024-06-01:
Hello,

A small hint on reporting the error. If - at school - you take a picture of the error, then you can later paste the photo in this (or another) issue.

Sorry for the long post 😃 

On a new issue it is asked to report some status, like ```uname -r```

On my ev3 they report for `uname -r`:
```
4.14.117-ev3dev-2.3.5-ev3
```
And for `dpkg-query -l {python3,micropython}-ev3dev*`:
```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                 Version                 Architecture            Description
+++-====================================-=======================-=======================-==============================================================================
ii  micropython-ev3dev2                  2.1.0                   all                     Python language bindings for ev3dev for MicroPython
ii  python3-ev3dev                       1.2.0                   all                     Python language bindings for ev3dev
ii  python3-ev3dev2                      2.1.0                   all                     Python language bindings for ev3dev
```

I do not have all sensors in the program, nor the model, so cannot test.
1. But running a part of the code, I see 
```
AttributeError: 'DriveBase' object has no attribute 'brake'
```

2. Another problem might be to **stop** a motor that is part in a DriveBase?

A short program that does not use a DriveBase has no problem:
```python
#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import Motor
from pybricks.parameters import Port, Stop
from pybricks.tools import wait
# from pybricks.robotics import DriveBase

# Create your objects here.
ev3 = EV3Brick()
# left_motor = Motor(Port.A)
right = Motor(Port.D)
# robot = DriveBase(left_motor,right,wheel_diameter=55.5, axle_track=104)
# robot.drive(100,0)

print("start motor")
right.run(50)
wait(2000)

try:
    print("\nright.hold()")
    right.hold()
except Exception as e:
    print("exception on hold()", e)

try:
    print("\nright.brake()")
    right.brake()
except Exception as e:
    print("exception on brake()", e)

try:
    print("\nright.stop()")
    right.stop()
except Exception as e:
    print("exception on stop()", e)
finally:
    right.run(0)  # try if this does stop the motor
    wait(10)

print("program at the end")
```

Program using the DriveBase gets permission violation if stop on a motor is done:
```python

#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import Motor
from pybricks.parameters import Port, Stop
from pybricks.tools import wait
from pybricks.robotics import DriveBase

# Create your objects here.
ev3 = EV3Brick()
left_motor = Motor(Port.A)
right = Motor(Port.D)
robot = DriveBase(left_motor,right,wheel_diameter=55.5, axle_track=104)
robot.drive(100,0)

print("start motor")
right.run(50)
wait(2000)

try:
    print("\nright.hold()")
    right.hold()
except Exception as e:
    print("exception on hold()", e)

try:
    print("\nright.brake()")
    right.brake()
except Exception as e:
    print("exception on brake()", e)

try:
    print("\nright.stop()")
    right.stop()
except Exception as e:
    print("exception on stop()", e)
finally:
    right.run(0)  # try if this does stop the motor
    wait(10)

print("program at the end")
```

The error:
```
# ./issue_1631_with_drivebase.py
start motor
Traceback (most recent call last):
  File "./issue_1631_with_drivebase.py", line 16, in <module>
OSError: [Errno 1] EPERM:

The requested operation is not valid in the current state:
--> Check the documentation for required conditions.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.

```

3. Note to pybricks team: I have to run these programs as root, as my personal user states:
```
Could not initialize graphics. Be sure to run using `brickrun -r -- pybricks-micropython`.
```
and that fails. But that is no issue for me.

Bert

BertLindeman, 2024-06-01:
Some extra info:
```'technichub', '3.5.0', 'v1.20.0-23-g6c633a8dd on 2024-04-11' ``` does not complain about the motor stop while it is in use as part of a DriveBase.

A beta I downloaded once to the EV3 ``` pybricks-micropython3.3.0b9``` also does not complain about this.

Although I think it strange to have a DriveBase and then manipulate one of the drives....

Hope this helps a bit.

Bert

Robocup-Programmer-QASMT, 2024-06-01:
Hello! The error that is occurring is the 3rd one. Why is it happening and how can I fix it please? It is happening at line 19 and several other times throughout the code

BertLindeman, 2024-06-02:
Hello,

And another **long** post. Can't help it.

I am confused by your answer.
 You write: 
 >  The error that is occurring is the 3rd one. 
...
 > It is happening at line 19

But the program **does not run at all** in case of this message:
``` Could not initialize graphics. Be sure to run using `brickrun -r -- pybricks-micropython`.```

I do not know how to fix that.

***

On the error ```AttributeError: 'DriveBase' object has no attribute 'brake'```
Take a look at [the ev3doc](https://pybricks.com/ev3-micropython/robotics.html)  and see there is no brake, but you could use 
```python
robot.stop()
```

***

The ```EPERM``` problem I did not get running a newer pybricks level. But is that allowed in your competition?

I downloaded [pybricks-micropython from the builds](https://github.com/pybricks/pybricks-micropython/actions/runs/8556042869)
and saved it on the EV3 as ```/usr/local/bin/pybricks-micropython_build_3357```

This version is reported as ```'3.5.0b1', 'ci-build-3357-v3.5.0b1-12-gd3f38000 on 2024-04-04')```

Log from the original pybricks-micropython and the build-3357 one:

### pybricks-micropython ./issue_1631_with_drivebase.py
```
start robot.drive()
start right motor
Traceback (most recent call last):
  File "./issue_1631_with_drivebase.py", line 17, in <module>
OSError: [Errno 1] EPERM:

The requested operation is not valid in the current state:
--> Check the documentation for required conditions.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.
```

### pybricks-micropython_build_3357 ./issue_1631_with_drivebase.py
```
start robot.drive()
start right motor
program at the end
```

<details> <summary>The test source</summary>

```python

#!/usr/bin/env pybricks-micropython
# from pybricks.hubs import EV3Brick
from pybricks.ev3devices import Motor
from pybricks.parameters import Port, Stop
from pybricks.tools import wait
from pybricks.robotics import DriveBase

# ev3 = EV3Brick()
left_motor = Motor(Port.A)
right = Motor(Port.D)
robot = DriveBase(left_motor,right,wheel_diameter=55.5, axle_track=104)
print("start robot.drive()")
robot.drive(100,0)
wait(1000)

print("start right motor")
right.run(50)
wait(1000)

try:
    right.hold()
except Exception as e:
    print("exception on hold()", e)

try:
    right.brake()
except Exception as e:
    print("exception on right.brake()", e)

try:
    right.stop()
except Exception as e:
    print("exception on stop()", e)

try:
    robot.brake()
except Exception as e:
    print("exception on robot.brake()", e)

finally:
    robot.stop()
    wait(1000)

robot.stop()
left_motor.stop()
right.stop()
print("program at the end")

```
</details>

***

Not sure where to find the documentation that comes along with ci-build-3357.
Maybe the pybricks team can add a more solid help here 😁 
Any info about EV3 python version allowed on Robocup Junior competition?

Bert

Robocup-Programmer-QASMT, 2024-06-02:
All are allowed

The test source

Robocup-Programmer-QASMT, 2024-06-02:
Due to time constraints I have switched over to block coding the bot 😢 however will be back to python for the next comp.

BertLindeman, 2024-06-02:
Side remark: pybricks 3.5 doc shows there **is** a drivebase brake() now.

Robocup-Programmer-QASMT, 2024-06-02:
Thank you for your assistance in resolving this issue. I have decided to make the switch from micro python to ev3 dev python and it is working perfectly now on very similar code. I appreciate your willingness to help