ev3dev/ev3dev #1635: Python 3.11 venv available -- Try out your favorite programs & run test programs for newer python features with the image at Discussion #1634!

Growflavor, 2024-06-17:
A Python 3.11 venv is included in a Debian 10 "Buster" experimental image: details at Discussion #1634

This image has worked well so far & is currently the 'goto' EV3dev image that my son & I use for all our EV3 projects...

Try out your favorite programs & run test programs to evaluate the 'usability' of python features introduced since python 3.5 in the Python 3.11 venv -- all that is needed is to change the shebang at the start of your python file to:

#!/home/robot/py11venv/bin/python3.11

Please reply to Discussion #1634 with your results (& link to your code if available online)!

My son & I look forward to your reports on what you can make python 3.11 do (or not do ;-) on the EV3...!

Nikolay-Zagrebin, 2024-06-17:
> A Python 3.11 venv is included in a Debian 10 "Buster" experimental image: details at Discussion #1634
>
> This image has worked well so far & is currently the 'goto' EV3dev image that my son & I use [...]

I'm following your posts, thank you so much for your work. Everything is very interesting.
I decided to build my image on Bullseye, but I can't install Brickman 0.11 from StepanTheFlowey correctly.

I do everything according to the instructions, the whole installation goes fine, but after flashing the SD card, the EV3 itself has an interface with unreadable text (((
Is there anything else undocumented?

There is a desire to test approximately similar builds on Bullseye and Stretch

Nikolay-Zagrebin, 2024-06-19:
I abandoned Bullseye, ev3 is running very slowly on it.
But I managed to succeed with Stretch, using docker I managed to install Python 3.12.4, I like the speed of work. I'm experimenting with installing the rest of the packages (numpy, scipy..)

Growflavor, 2024-06-20:
> managed to succeed with Stretch, using docker I managed to install Python 3.12.4, I like the speed of work. I'm experimenting with installing the rest of the packages (numpy, scipy..)

1) Did you altinstall 3.12.4 so that Stretch runs on 3.5 & we can run 3.12.4 as venv?

2) I also found the performance improvements introduced starting in Python 3.11 to be very significant on the EV3...

about 100 MORE loops per second with the [EV3fast.py test](https://github.com/QuirkyCort/ev3dev-lang-python-fast) !  will post details at #1634 

3) SciPy appears to need to be cross-compiled since one of its build dependencies openBLAS has no package for armel...the build configure does not use the armel libBLAS...

I am posting EV3 wheels that pip builds in python 3.11.9 to:

https://github.com/Growflavor/EV3projects/tree/main/PIPwheelsEV3

Nikolay-Zagrebin, 2024-06-21:
> 1. Did you altinstall 3.12.4 so that Stretch runs on 3.5 & we can run 3.12.4 as venv?

To install, I used the docker image ev3dev/ev3dev-stretch-ev3-base.
I'm new to this, as always something went wrong)). For example, in ev3 itself, the pip list -v command triggers the "Illegal instruction" message.
https://drive.google.com/drive/folders/1THkrEt4HaVb3WftCPcanT-3_YIArbgAp?usp=sharing

> 2\. I also found the performance improvements introduced starting in Python 3.11 to be very significant on the EV3...

Interesting

> 3\. SciPy appears to need to be cross-compiled since one of its build dependencies openBLAS has no package for armel...the build configure does not use the armel libBLAS...
>     https://docs.scipy.org/doc/scipy/building/index.html

Yes, it's with libBLAS that I'm stuck.

Nikolay-Zagrebin, 2024-06-28:
> about 100 MORE loops per second 3.11 vs 3.7 with the [EV3fast.py test](https://github.com/QuirkyCort/ev3dev-lang-python-fast) ! will post details at #1634

I managed to install Python 3.12.4 on EV3, though bypassing docker.
I have completed performance testing based on the tests
https://github.com/QuirkyCort/ev3dev-lang-python-fast?tab=readme-ov-file

All images have the following software
robot@ev3dev:~$ pip list -v
Package        Version     Location                          Installer
-------------- ----------- --------------------------------- ---------
Cython         0.29.37     /usr/lib/python3.11/site-packages pip
pillow         10.3.0      /usr/lib/python3.11/site-packages pip
pip            24.1        /usr/lib/python3.11/site-packages pip
plumbum        1.8.3       /usr/lib/python3.11/site-packages pip
python-ev3dev  1.2.0       /usr/lib/python3.11/site-packages pip
python-ev3dev2 2.1.0.post1 /usr/lib/python3.11/site-packages pip
rpyc           6.0.0       /usr/lib/python3.11/site-packages pip
setuptools     59.2.0      /usr/lib/python3.11/site-packages pip
wheel          0.37.0      /usr/lib/python3.11/site-packages pip

My results
Line Follower Benchmark			
Python on EV3	ev3dev	ev3fast	ev3dev2
3.12.4	                 39.51	240.77	48.85
**3.11.9	                 43.68	258.96	51.87**
3.10.9	                 32.33	170.69	36.54
3.9.19	                 31.45	159.21	35.46
3.5.3	                         24.85	139.06	28.20

Growflavor, 2024-07-02:
@Nikolay-Zagrebin
While not yet leveraging the benefit of python 3.11 due to issues with building scipy from source, 
& while likely too slow to be useful except in limited cases, still could be fun for limited basic learning,

via docker container, using the system scipy packages:
 
python3 -m venv --system-site-packages venv37

pip does succeed installing the robotics toolbox (requires python >=3.6) 

for both the
Debian 10 Buster python 3.7 venv [link to "Buster" tinker image .zip with venv37](https://drive.google.com/file/d/1CTgfor0x5ChCO9ZSYRA7_vibquguLeqN/view?usp=drive_link)
SHA-256: 3AC8FE05C2D0CAEAA6FA7586FCA97E5AC183A51409C12CD9AA3087153C0C4A43
Debian 11 Bullseye python 3.9 venv 
(Bulleye on 4.19 kernel image is a bit slower than the Buster image but works OK for me once file system is expanded to use the whole 32GB sdcard) 

https://github.com/jhavl/spatialmathematics
https://github.com/petercorke/robotics-toolbox-python

robot@56df5ed72b53:~$ source venv37/bin/activate
(venv37) robot@56df5ed72b53:~$ python3 -m pip install roboticstoolbox-python sympy

<details>
  <summary>Successfully built roboticstoolbox-python spatialgeometry swift-sim rtb-data progress</summary>
 
Installing collected packages: spatialgeometry, pgraph-python, websockets, swift-sim, rtb-data, progress, roboticstoolbox-python, mpmath, sympy

Successfully installed mpmath-1.3.0 pgraph-python-0.6.2 progress-1.6 roboticstoolbox-python-1.1.1 rtb-data-1.0.1 spatialgeometry-1.1.0 swift-sim-1.1.0 sympy-1.10.1 websockets-11.0.3
 
  ```javascript

(venv37) robot@56df5ed72b53:~$ pip list
Package                Version
---------------------- -----------
agt                    1.0
ansitable              0.10.0
asn1crypto             0.24.0
atomicwrites           1.1.5
attrs                  18.2.0
beautifulsoup4         4.7.1
cfgv                   3.3.1
chardet                3.0.4
colored                1.4.4
cryptography           2.6.1
cycler                 0.10.0
decorator              4.3.0
distlib                0.3.8
entrypoints            0.3
evdev                  1.1.2
filelock               3.12.2
html5lib               1.0.1
identify               2.5.24
importlib-metadata     6.7.0
joblib                 0.13.0
keyring                17.1.1
keyrings.alt           3.1.1
kiwisolver             1.0.1
lxml                   4.3.2
matplotlib             3.0.2
more-itertools         4.2.0
mpmath                 1.3.0
nodeenv                1.9.1
nose                   1.3.7
numexpr                2.6.9
numpy                  1.21.6
openrobertalab         1.7.4
pandas                 0.23.3+dfsg
pgraph-python          0.6.2
Pillow                 5.4.1
pip                    18.1
[... 99 lines cut ...]
tmpfs            20M  324K   20M   2% /run
/dev/mmcblk0p2   30G  4.0G   25G  15% /
tmpfs            28M     0   28M   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs            28M     0   28M   0% /sys/fs/cgroup
/dev/mmcblk0p1   48M  230K   48M   1% /boot/flash
tmpfs           5.6M     0  5.6M   0% /run/user/1000

robot@ev3dev:$ free -h
              total        used        free      shared  buff/cache   available
Mem:           55Mi        28Mi       1.0Mi       0.0Ki        26Mi        23Mi
Swap:         607Mi        26Mi       581Mi

robot@ev3dev:$ sudo systemctl stop zram_swap.service

robot@ev3dev:$ free -h
              total        used        free      shared  buff/cache   available
Mem:           55Mi        32Mi       1.0Mi       0.0Ki        21Mi        19Mi
Swap:         511Mi        61Mi       450Mi
robot@ev3dev:~$ cat /proc/swaps
Filename                                Type            Size    Used    Priority
/swapfile                               file            524284  63172   1
robot@ev3dev:~$ `
  ```  
</details>

dlech, 2024-07-02:
It's being used right now.

```console
$ cat /proc/swaps 
Filename				Type		Size	Used	Priority
/dev/zram0                              partition	98300	18432	16383
$ free
              total        used        free      shared  buff/cache   available
Mem:          57620       22992        1308         144       33320       30816
Swap:         98300       18176       80124
```

Maybe you broke the `zram_swap.service` in your custom images?

If you are doing things that require more RAM than that, you should upgrade to something more powerful like a Raspberry Pi, IMHO. Using swap on the SD card is super slow and wears out the SD card faster.

Growflavor, 2024-07-03:
> It's being used right now.

I do expect that we will break & learn how to fix many things over the next few years.  Yes, by itself the 98300 zram swap works fine...I was referring to a similar zram swap + swapfile combination [observation in #571](https://github.com/ev3dev/ev3dev/issues/571#issuecomment-196514303) & your recommendation (that indeed works well) to that other person back in 2016  "Hmm... the zram swap is actually just compressed ram, so since it is full, it could be filling up your actual ram. Try turning off the zram swap and just using your swapfile."

> If you are doing things that require more RAM than that, you should upgrade to something more powerful like a Raspberry Pi,

Yes, indeed when we get to actual physical robotics projects that would use such tools.  What is occurring now, is that at age 12, the environment that you have been a key part of creating with the EV3dev ecosystem has spurred an unprecedented level of self driven learning by my son about _how_ packages get built & how these systems work...so yes, some things we do will appear impractical from a performance perspective, but are being done because they are for whatever reason inspiring/driving a remarkable interest in learning. 

Yesterday, my son asked me if I would help him begin to study your kernel work this summer.  I am not yet familiar with kernels at that level and would be grateful for your recommendation of what resources you assess are best/most relevant to start to begin to understand the EV3 drivers in the kernel, kernel configuration, etc. as this may well become a regular study project over the next 5 years...it may seem unusual even to me but then he did quickly become better than me in autodesk fusion 360 at age 9 and decided to teach himself to solve any pattern of 3x3 cubes in under 40 seconds in a matter of days...(Mindcub3r was actually what led me to discover the EV3 & EV3dev).  Thank you.

dlech, 2024-07-03:
> "Hmm... the zram swap is actually just compressed ram, so since it is full,

There is a `Priority` column in the output of `/proc/swaps`, so maybe it would be possible to make the zram priority less that the SD card, then you wouldn't have to disable the zram swap?

> some things we do with the EV3 will appear impractical from a performance perspective, but are being done because they are inspiring

:sunglasses: 

> what resources you assess are best/most relevant to start to begin to understand the EV3 drivers in the kernel, kernel configuration,

One place to start would be to just build the kernel. https://github.com/ev3dev/ev3dev-buildscripts

Since there kernel is constantly evolving, the best way to learn about it is just be reading the kernel source.

The latest documentation build from source is also online. Since drivers are the most relevant to ev3dev, here is a good place to start: https://www.kernel.org/doc/html/latest/driver-api/driver-model/index.html

Bootlin has some free training materials: https://bootlin.com/training/kernel/ (actual course requires payment, but slides are freely available and contain a nice higher-level overview of the kernel focusing on embedded systems)

Growflavor, 2025-01-13:
> > about 100 MORE loops per second 3.11 vs 3.7 with the [EV3fast.py test](https://github.com/QuirkyCort/ev3dev-lang-python-fast) ! will post details at #1634
> 
> My results [image]
> 
> It can be seen that 3.11.9 shows excellent results

@Nikolay-Zagrebin 
Last night I came across @dlech 's post & applied the 456 MHz setting to one of our EV3 bricks:

https://lechnology.com/2018/06/overclocking-lego-mindstorms-ev3-part-2/
lscpu | grep Bogo
sudo nano /boot/flash/uEnv.txt
in a SSH terminal on your EV3. Remove the comment (the #) from the line that says cpufreq=375
I set cpufreq=456

<details>
  <summary>(Click the black triangle to reveal details) Gives us an increase of additional 100 loops per second :-)
</summary>
 
Debian Buster Python 3.7 = #!/usr/bin/env python3

  ```javascript
 Starting: brickrun -r --directory="/home/robot/CAL" "/home/robot/CAL/EV3devFastTest.py"
----------
259.4326294474275
>>> exit()

Starting: brickrun -r --directory="/home/robot/CAL" "/home/robot/CAL/EV3devFastTest.py"
----------
256.276079388951
>>> 

Starting: brickrun -r --directory="/home/robot/CAL" "/home/robot/CAL/EV3devFastTest.py"
----------
276.44629172193686
>>> 
  ```

Debian Buster Python 3.11 venv = #!/home/robot/venv311/bin/python3.11

  ```javascript
Starting: brickrun -r --directory="/home/robot/CAL" "/home/robot/CAL/EV3devFastTestVenv311.py"
----------
363.82051595483824
>>> 

Starting: brickrun -r --directory="/home/robot/CAL" "/home/robot/CAL/EV3devFastTestVenv311.py"
----------
347.29525045411793
>>> 

Starting: brickrun -r --directory="/home/robot/CAL" "/home/robot/CAL/EV3devFastTestVenv311.py"
----------
367.2200962032974
>>> 
  ```
</details>

dlech, 2025-01-13:
> since I've been preoccupied

I wish you the best and hope you can find the right doctors and treatment that you need. And thanks for your contributions to ev3dev so far. I hope you will be able to get well and continue the collaboration.