ev3dev/ev3dev #1637: How to configure ev3dev USB SSH in FreeBSD

7joaojari7, 2024-06-25:
Hi, i need to use ev3dev but i am having trouble connecting through usb cable in freebsd.
I am a newbie in ev3dev and BSD in general, please help me. Switching to Windows or Linux is not a option.
thanks

dlech, 2024-06-25:
Sorry, don't know much about freebsd, but ev3dev uses standard network over USB interfaces. The zeroconf name resolution might not work if Avahi isn't available in freebsd. But after you connect the USB cable, there should be an IP address that appears on the EV3 that you can use in your SSH command to connect.

Growflavor, 2024-07-03:
> Hi, i need to use ev3dev but i am having trouble connecting through usb cable in freebsd. 

EV3's USB A port: 
When possible, I suggest using the USB A port with an USB to ethernet adapter to connect with the EV3.   This way of connecting to the EV3 has been fast & very reliable for us on all 5 of our EV3 bricks...this way also enables having all bricks connected to their own VScode windows/environment simultaneously on the same PC...

Looks like VScode has been ported to freebsd:
https://freebsdfoundation.org/resource/how-to-use-vs-code-on-freebsd/

EV3's USB mini port: 
I also do not know about freebsd, but can say that even with Windows 10/11, for us the EV3's USB mini port is 'limited'...tends to work OK for 'no IP address' access via VScode EV3dev browser or ssh to 'ev3dev' port 22.... & 'sometimes' we can get a routable IP connection via mini USB port if we have Windows Internet Connection Sharing ICS active...