ev3dev/ev3dev #1638: Update APT sources for `docker-cross` images.

monomycelium, 2024-06-28:
**Describe the bug**
I was trying to build the `debian-stretch-cross` image for [`docker-cross`](https://github.com/ev3dev/docker-cross) (because they only existed for `x86_64`), when the `apt-get update` command kept running into issues. Apparently, official mirrors for older releases of Debian (which now includes "stretch") are hosted at `http://archive.debian.org`. After updating the outdated [mirrorlist](https://github.com/ev3dev/docker-cross/blob/master/ev3dev-stretch/apt.sources.list.debian) using [a more recent one](https://github.com/ev3dev/docker-library/blob/master/ev3dev-stretch/layers/debian/etc/apt/sources.list), I got strange dependency errors when installing packages:
``` console
# apt install -y cmake git wget
The following packages have unmet dependencies:
 cmake : Depends: libcurl3 (>= 7.16.2) but it is not going to be installed
 git : Depends: libcurl3-gnutls (>= 7.16.2) but it is not going to be installed
 wget : Depends: libgnutls30 (>= 3.5.6) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
I finally updated it to the following (thanks to [DebianSupport](https://debiansupport.com/mirrors/)), and the image was successfully built.
```
deb http://archive.debian.org/debian/ stretch main contrib non-free
deb http://archive.debian.org/debian/ stretch-proposed-updates main contrib non-free
deb http://archive.debian.org/debian-security stretch/updates main contrib non-free
```

**To Reproduce**
Steps to reproduce the behaviour:
1. [Get Docker](https://docs.docker.com/get-docker/).
2. Execute the following in a shell:
``` console
$ git clone https://github.com/ev3dev/docker-cross.git
$ cd ev3dev-stretch
$ docker image build --rm -t ev3 -f './debian-stretch-cross.dockerfile' .
```
3. Check for similar errors.

**Expected behaviour**
Packages should install without errors and image should be successfully built.

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04.4 LTS (Virtualised)
 - Connection Type: Virtual Ethernet
 - Other software: (null)

**Additional context**
The `docker-cross` image by `ev3dev` can help cross-compile software for the robot inside a container with little overhead but great reproducibility.

dlech, 2024-06-28:
Could you make a pull request to fix this?

monomycelium, 2024-06-30:
While testing for Jessie, I found that the Debian Cross-toolchains repository at http://emdebian.org/tools/debian/ is down. [This](https://wiki.debian.org/CrossToolchains#In_jessie_.28Debian_8.29) Debian wiki still refers to the repository, and so does the `debian-jessie-cross` Dockerfile. Removing the repository from the sources list causes the following error:
``` console
$ apt install crossbuild-essential-armel crossbuild-essential-armhf ...
...
E: Unable to locate package crossbuild-essential-armel
E: Unable to locate package crossbuild-essential-armhf
```

monomycelium, 2024-06-30:
I found [this mail thread](https://www.mail-archive.com/debian-embedded@lists.debian.org/msg07791.html) about it, and apparently, the repository can now be accessed at http://emdebian.metrotek.center/. When I tried changing the sources (and having to deal with HTTPS and missing keys), I get the following error:
``` console
$ apt-get install crossbuild-essential-armel crossbuild-essential-armhf ...
...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 crossbuild-essential-armel : Depends: gcc-arm-linux-gnueabi (>= 4.9.1-1) but it is not going to be installed
                              Depends: g++-arm-linux-gnueabi (>= 4.9.1-1) but it is not going to be installed
 crossbuild-essential-armhf : Depends: gcc-arm-linux-gnueabihf (>= 4.9.1-1) but it is not going to be installed
                              Depends: g++-arm-linux-gnueabihf (>= 4.9.1-1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

dlech, 2024-06-30:
I'd rather not spend time trying to make jessie work. It was never really finished in the first place.