ev3dev/ev3dev #1642: Adding sensor dirvers for HX7111

igor-koop, 2024-07-17:
I am trying to implement a weight sensor based on a load cell and a HX711 analog-to-digital converter. HX711 uses a non-i2c compliant two wire protocol which makes it necessary to implement new drivers for this sensor. As per the documentation I am opening this issue to hopefully recieve some technical support/guidance on achieving my goal.

dlech, 2024-07-17:
I assume this is the right datasheet? https://www.digikey.com/htmldatasheets/production/1836471/0/0/1/hx711.html

This looks like you would need a SPI controller, but the EV3 doesn't have any external SPI connections. So your best bet would probably be to add a microcontroller between the load cell and the EV3 that reads the data from the load cell and converts it to I2C or UART that the EV3 can handle on the input ports.

igor-koop, 2024-07-17:
Thank you for your prompt response. I am a noobie and didn't realize that it's just impossible to establish direct communication between these parts. Will changing the converter from HX711 to NAU7802 (datasheet: https://cdn.sparkfun.com/assets/e/c/8/7/6/NAU7802_Data_Sheet_V1.7.pdf) work without any further intermediates? It should support conventional I2C, but want to make sure.

dlech, 2024-07-17:
That should have a better chance of working since it is I2C. One thing to note is that then 10k pullup resistors can sometimes be problematic and you might have to replace them with 82k.

igor-koop, 2024-07-31:
Update on my side: NAU7802 indeed works, but only if 10K resistors are replaced with 82K ones. I also attach a simple Python script with a class to interface the weight sensor. I am not quite sure how to start on the proper drivers.

```
#!/usr/bin/env python3

import smbus
import time
from typing import Optional, Union, List, Callable
from ev3dev2.sensor import INPUT_1, INPUT_2, INPUT_3, INPUT_4

class WeightSensor:
    """
    A class representing a load cell based weight sensor interfaced via NAU7802.

    The WeightSensor class provides methods and attributes to interact with a load cell based weight sensor
    that is connected to the NAU7802 ADC (Analog-to-Digital Converter) chip. The NAU7802 chip is a 24-bit dual-channel
    ADC specifically designed for bridge sensors, making it suitable for load cell applications.

    This class allows you to read weight measurements from the load cell, perform calibration, and configure various
    settings of the NAU7802 chip.
    """

    ##################################################
    #                   ATTRIBUTES                   #
    ##################################################

    # I2C address of the NAU7802
    # Nuvoton. (2012, January). NAU7802 datasheet: 24-Bit Dual-Channel ADC For Bridge Sensors, p. 16.
    __slave_address = 0x2A

    # NAU7802 register map
    # Nuvoton. (2012, January). NAU7802 datasheet: 24-Bit Dual-Channel ADC For Bridge Sensors, p. 28.
    __slave_registers = {
        'PU_CTRL':      0x00,   # Power up control                                                                              (Default = 0x00)
        'CTRL1':        0x01,   # Control 1                                                                                     (Default = 0x00)
        'CTRL2':        0x02,   # Control 2                                                                                     (Default = 0x00)
        'OCAL1_B2':     0x03,   # Offset calibration (Channel 1, Byte 2)                                                        (Default = 0x00)
        'OCAL1_B1':     0x04,   # Offset calibration (Channel 1, Byte 1)                                                        (Default = 0x00)
        'OCAL1_B0':     0x05,   # Offset calibration (Channel 1, Byte 0)                                                        (Default = 0x00)
        'GCAL1_B3':     0x06,   # Gain calibration (Channel 1, Byte 3)                                                          (Default = 0x00)
        'GCAL1_B2':     0x07,   # Gain calibration (Channel 1, Byte 2)                                                          (Default = 0x80)
        'GCAL1_B1':     0x08,   # Gain calibration (Channel 1, Byte 1)                                                          (Default = 0x00)
        'GCAL1_B0':     0x09,   # Gain calibration (Channel 1, Byte 0)                                                          (Default = 0x00)
        'OCAL2_B2':     0x0A,   # Offset calibration (Channel 2, Byte 2)                                                        (Default = 0x00)
        'OCAL2_B1':     0x0B,   # Offset calibration (Channel 2, Byte 1)                                                        (Default = 0x00)
        'OCAL2_B0':     0x0C,   # Offset calibration (Channel 2, Byte 0)                                                        (Default = 0x00)
        'GCAL2_B3':     0x0D,   # Gain calibration (Channel 2, Byte 3)                                                          (Default = 0x00)
        'GCAL2_B2':     0x0E,   # Gain calibration (Channel 2, Byte 2)                                                          (Default = 0x80)
        'GCAL2_B1':     0x0F,   # Gain calibration (Channel 2, Byte 1)                                                          (Default = 0x00)
        'GCAL2_B0':     0x10,   # Gain calibration (Channel 2, Byte 0)                                                          (Default = 0x00)
        'I2C_CONTROL':  0x11,   # I2C control                                                                                   (Default = 0x00)
        'ADCO_B2':      0x12,   # ADC conversion result (Byte 2, Read only)                                                     (Default = 0x00)
        'ADCO_B1':      0x13,   # ADC conversion result (Byte 1, Read only)                                                     (Default = 0x00)
        'ADCO_B0':      0x14,   # ADC conversion result (Byte 0, Read only)                                                     (Default = 0x00)
        'ADC_REG':      0x15,   # ADC register (Read only)                                                                      (Default = 0x00)
        'OTP_B2':       0x15,   # One-time programmable memory (Byte 2, Read only)                                              (Default = 0x00)
        'OTP_B1':       0x16,   # One-time programmable memory (Byte 1, Read only)                                              (Default = 0x00)
        'OTP_B0':       0x17,   # One-time programmable memory (Byte 0, Read only)                                              (Default = 0x00)
        'RES_00':       0x18,   # Reserved                                                                                      (Default = 0x00)
        'RES_01':       0x19,   # Reserved                                                                                      (Default = 0x00)
[... cut ...]
            weight = self.slope * raw + self.zero
            if self.tare is not None:
                weight -= self.tare
            return weight
        else:
            return None

    def tare(self) -> None:
        """
        Tare the sensor.

        :return: None
        """
        self.tare = self.average(100, self.get_weight)

if __name__ == '__main__':
    sensor = WeightSensor(INPUT_1)
    with sensor:
        print(sensor.calibrate('two-point', [0, 100]))
        print(sensor.get_weight())
```

dlech, 2024-08-01:
Wow, this looks great!