ev3dev/ev3dev #1650: ev3 not booting

Nissemanen, 2024-10-05:
**Describe the bug**
ev3 not booting. it just gets stuck at the "Mindstorms Starting" screen. im using a SanDisk ultra miroSDHC 32GB (they didnt have anything lower price nor storage)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://www.ev3dev.org/docs/getting-started/
2. Click on the button that says "Download ev3dev-stretch for LEGO MINDSTORMS EV3"
3. Go to https://etcher.balena.io
4. Scroll down to Downloads and download "Etcher for Macos x64" (if you dont see it refresh the page)
5. Start Balena Etcher app
6. Select the .img file then select the right disk
7. flash the img to the drive (then there being the files "boot.scr" "u-boot.bin" and "uEnv.txt"
8. put the microSDHC in the turned of ev3
9. try to start it
10. See error

**Expected behavior**
what should have happened was to after around 2 minuets it shuld have started with the splash text "ev3DevOrg" on the screen. but instead I'm greeted with 10 minuets of "Mindstorms starting"

**Screenshots**
If you need pictures just ask

**Desktop (please complete the following information):**
 - OS: macOS Monterey version 12.7.6
 - Connection Type: none right now but will use USB or WiFi
 - Other software: none right now but will use visual studio code

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: (i dont know where to look so ill just write te .img file name) ev3dev-stretch-ev3-generic-2020-04-10.img

**Additional context**
i got my ev3 Mindstorms set around 2016 i think

dlech, 2024-10-05:
There is a know issue with newer versions of Etcher corrupting the SD card. 

Workarounds include:
- unziping the image first and flashing the .bin file
- using Etcher v1.17.x
- using a different flasher program instead of Etcher

Nissemanen, 2024-10-06:
thank you! it now works as expected👍.

closed 2024-10-06 by Nissemanen