ev3dev/ev3dev #1651: using pyserial with the EV3

Nissemanen, 2024-10-08:
**Question**
How do you use pyserial to communicate with your computer? (if it even is possible)

**Background Information**
I am trying to use pyserial to communicate with my pc because I want to build a robot that sends over the angle of a motor to my pc. right now tough i just want to send over a changing message to test the connection and speed

code:
(sender / ev3)
```
#!/usr/bin/env python3
import serial
import time

ser = serial.Serial('/dev/ttyGS0', 115200)

message1 = "Hello World!"
message2 = "Hello from EV3!"
curmes = 1

while True:
    if curmes == 1:
        ser.write(message1.encode('utf-8'))
        print("message1 sent", file=sys.stderr)
        curmes = 2
    else:
        ser.write(message2.encode('utf-8'))
        print("message2 sent", file=sys.stderr)
        curmes = 1
    
    time.sleep(1)
```
(receiver / my pc)
```
import serial

ser = serial.Serial('/dev/COM1', 11520)

while True:
    if ser.in_waiting > 0:
        data = ser.readline().decode('utf-8').strip()
        print(f'Recived: {data}')
```

**Desktop:**
 - OS and version: Window 10 19045.4894
 - Connection Type: USB
 - Software: Visual Studio Code

**Robot:**
 - Device: [e.g. EV3, BrickPi3]
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165353A81F  
BOARD_INFO_TYPE=main
```

dlech, 2024-10-08:
Ev3dev isn't set up to do USB serial by default. You would have to modify the usb gadget setup script to add it.

Or, you can just use a network socket instead of serial since ev3dev USB provides a network connection by default.

Nissemanen, 2024-10-09:
oh ok. so im guesing that network socet means to use either bluetooth or wifi to connect to the computer and send data? or am i getting it wrong. im very new to this type of python programing and just curius. if you have any links please send them. thank you in advance!

dlech, 2024-10-09:
> so im guesing that network socet means to use either bluetooth or wifi to connect to the computer and send data?

No, there is also network over USB.

Nissemanen, 2024-10-09:
ok i'll look in to it. thank you for the help i realy appreciate it

Nissemanen, 2024-10-10:
Wait

closed 2024-10-10 by Nissemanen