ev3dev/ev3dev #1658: Mindsensors "8 Channel Servo Controller for NXT or EV3"

Beisshase, 2024-12-21:
Hello,

I have am "8 Channel Servo Controller for NXT or EV3" from Mindsensors, but I can not get it to work.
I am using ev3dev with an EV3 and Micropython.

The documentaion I have read did not give me a clue how to connect to this device.

Can somebodyhelp me with some lines of code to show me how I connect und give some basic instructions to move this thing?

Thanks

dlech, 2024-12-22:
There aren't any premade classes for controlling servo motors like there are for other kinds of motors, so you will need to access the sysfs files directly to be able to use it.

Docs the ev3dev servo motor drivers are here: http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/motors.html#servo-motor-subsystem

If you want to read the battery voltage from it, you can use [the Ev3devSensor class](https://pybricks.com/ev3-micropython/iodevices.html#pybricks.iodevices.Ev3devSensor) along with [this info
](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#ms-8ch-servo).

Beisshase, 2024-12-22:
Thanks.
Can you be more specific about what steps need to be taken to access the servos?
Do I have to 'Connect' it with a command in Sysfs?
Aftre that can I access it from python?

dlech, 2024-12-22:
You just read and write the files. The main ones are `position_sp` and `command`. For example, if you write `"run"` to the `command` file, the motor will run to the 0 position.

Beisshase, 2024-12-22:
ok, that will do the trick.
But the servo just "Ticks" a little and does not move to instructed position.
Also the content of position_sp stays at -150. But the timestamp of the file changes to the current time....?!

Growflavor, 2025-01-24:
> But the servo just "Ticks" a little and does not move to instructed position.

@Beisshase Thanks for bringing the [mindsensors "8 Channel Servo Controller for NXT or EV3" ](https://www.mindsensors.com/home/62-8-channel-servo-controller-for-nxt-or-ev3.html) to my attention...I am curious if you made any progress?

Before trying the micropython, did you happen to successfully test control of standard LEGO parts via the "8 channel servo controller" & EV3-G & the default blocks provided by mindsensors (ideally also sample programs they say they provide but I could not locate a simple one one so far...)?

This video has an interesting insight about finding out the servo 'start position' when controlling various servos with this controller (link starts at the part of the video that discusses this issue -- earlier in the video also discusses supplying bus voltage for you specific servo requirements):

https://youtu.be/f_dkbnj7afc?si=FRuYpyZXg9GGKi91&t=244

Also, I am curious what type of servo are you trying to control e.g. continuous vs 90 deg vs 180 deg (any links to components & photos of your setup to share :-)?

    Any RC servo motor with "1500 µS neutral" specifications.
    Use 90° or 180° or continuous rotation RC servos.
    Supports standard RC Servo connector plugs.
    Press-in terminals for battery.

From: https://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/motors.html#id24

    Used to set the pulse size in milliseconds for the signal that tells the servo to drive to the _mid position_sp_. 
    **Default value is 1500**. Valid values are 1000 to 2000. 
    **For example, on a 180 degree servo, this would be 90 degrees.** 
    **On continuous rotation servo, this is the “neutral” position where the motor does not turn.**

From: https://www.mindsensors.com/home/index.php?controller=attachment&id_attachment=107

Servo Position Register:

    The position (low byte/high byte) is a 16-bit number, which directly sets the output pulse width in uS.
    Setting the position to 1500 (1500uS or 1.5mS) will typically set servos to their center position.
    The range of pulse widths that are normally supported are from 500uS (0.5mS) to 2500uS (2.5mS).
        
    **Take care though, as it is easy to make the servo run into internal stops,
     if you give it pulse widths at the upper or lower extremes.**
    
    The registers can also be read back. The position will be the current position of the servo
    during a speed-controlled movement, so you can track its progress towards the requested position.
    Setting the servo position to 0 will deactivate the servo output rendering servo in floating condition.

As a project I might explore with my son, I see that the [LEGO block developer kit](https://education.lego.com/en-us/product-resources/mindstorms-ev3/downloads/developer-kits/) can reveal internals of the [custom block](https://www.mindsensors.com/home/index.php?controller=attachment&id_attachment=106) provided by mindsensors:

https://i.imgur.com/WZEcMIi.png

You have definitely set me down a rabbit hole of new possibilities ;-)

[sensor port splitter](https://www.mindsensors.com/home/50-port-splitter-for-nxt-digital-sensors.html)
    Q: Can I connect three NXTServo units to one Port Splitter?
    Ans:  Yes, you can. Ensure that your each NXTServo has unique I2C address.

one [eurobricks reference discussion](https://www.eurobricks.com/forum/index.php?/forums/topic/168465-ev3-vs-arduino-using-rc-servos/)
    
    https://youtu.be/N8iDY9oTMig?si=p9gaheV8mbNevR0C&t=160
    
    https://www.qikeasy.com/

    https://coresg.tech/evn/