ev3dev/ev3dev #1659: How to override/ignore EV3 Large Motor Max DPS

ntxngle, 2025-01-12:
Hello,
Very satisfied with ev3dev and the python bindings.
However, I am running the ev3 (and by extension the motor) at 12V, so the large motor (lego-ev3-l-motor) is capable of theoretically 1380 degrees per second, which can be verified with run-direct at duty_cycle_sp at 100. However, for commands like run-to-abs-pos, speed_sp is not allowed to exceed 1050 (the driver throws error), which is the driver's value of the maximum speed of the lego-ev3-l-motor.
Addtionally, SpeedPercent(100) also uses the 1050 value.
How can I go about ignoring or changing the driver's maximum value for DPS?
For example, is it possible to connect the (driver name) as a generic tacho motor which the driver cannot know the restrictions of?
Another potential solution is editing ev3_motor_defs.c in linux-lego-drivers, but I am unsure on how to rebuild.

Growflavor, 2025-01-13:
while I do not have insights for your question, (but hoping that @dlech might since I would enjoy reading his response),
I am curious about the running of the EV3 at 12V...do you happen to have a link to a description of how this is done (my searching the www so far did not find a useful link on the subject).  Thanks!

dlech, 2025-01-13:
While we love every EV3 hack here at ev3dev, we didn't anticipate this one (12V power). :smile: 

The max speed is [hard-coded in the driver](https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-buster/motors/ev3_motor_defs.c#L64) so the only way around it is to change the driver.

Or just use the `duty_cycle` attribute and make your own motor control algorithm to control the speed.

ntxngle, 2025-01-14:
@Growflavor 
I imagined it could be done when I read Philo's Home motor page (https://philohome.com/motors/motorcomp.htm)
[image]

A search confirmed it possible ([eurobricks post](https://www.eurobricks.com/forum/index.php?/forums/topic/195337-ev3-ac-power-adapter/&tab=comments#comment-3628298) and similar [philohome for nxt](https://philohome.com/nxtsupply/nxtsupply.htm))

I don't use a fancy adapter like he does, just alligator clips right onto the terminals (which unfortunately pops off frequently, much to fsck's dismay)

The voltage meter in the ev3 reports 10.8, but multi meter (and motor speed) confirms it truly is 12v.

Make sure to use switching adapter and triple check which battery terminals you connect to, other than that very simple mod, the easiest part so far lmao.

@dlech 

I though about using duty cycle to control but python is much to slow to get the control I need. The project is based entirely on run-to-abs-pos, which can't be implemented precisely with run_direct in python (or at least not with my mediocre python skills)

Yes, I found the motor defs but how would I go about rebuilding that kernel module after modification? Do I need to rebuild the entire ev3dev kernel as well? I read the ev3dev documentation for development but it remains unclear.

dlech, 2025-01-14:
> how would I go about rebuilding that kernel module

I don't think that one is compiled as a module, so you would need to build the full kernel. There is some info on how to do that at https://github.com/ev3dev/ev3dev-buildscripts

ntxngle, 2025-01-15:
Ok @Growflavor , I just built the deb with the patched motor defs, will test it soon

Growflavor, 2025-01-15:
> Ok @Growflavor , I just built the deb with the patched motor defs, will test it soon

Great!  I'd better get my act together ;-)  

I also want to build the kernel but with the CONFIG_ZRAM_WRITEBACK enabled in the kernel config...
which should enable the kernel to keep using the zram dlech configured but also when zram fills to use a swap file in combination with the zram... (useful if [playing with silly stuff on the 'slow' EV3](https://github.com/ev3dev/ev3dev/issues/1635#issuecomment-2204008717)  instead of just in colab/jupyter notebooks ;-) 

"# CONFIG_ZRAM_WRITEBACK" in kernel config 

"kernels since 4.14 should support # CONFIG_ZRAM_WRITEBACK so that swapfile will be used if ZRAM gets full"

ntxngle, 2025-01-15:
@Growflavor It worked!  I built the kernel from https://github.com/ev3dev/ev3dev-buildscripts in a ubuntu server lts 20 in virtualbox flawlessly. If you tell me which file to edit I can do so and build a copy for you, it's fairly small

closed 2025-01-15 by ntxngle

Growflavor, 2025-01-15:
> @Growflavor It worked! I built the kernel from https://github.com/ev3dev/ev3dev-buildscripts in a ubuntu server lts 20 in virtualbox flawlessly. If you tell me which file to edit I can do so and build a copy for you, it's fairly small

1) Fantastic! also, the kernel config to enable zram & swap should just be this one line:

[image]

2) did you build the default branch = Buster 4.19.85 kernel & if so, did you need to make any changes to the buildscripts -- I get a few errors when I try to build the default Buster branch:

 CALL    /home/growflavor/ev3d10kern/ev3-kernel/scripts/checksyscalls.sh
/usr/bin/ld: scripts/dtc/dtc-parser.tab.o:(.bss+0x10): multiple definition of `yylloc'; scripts/dtc/dtc-lexer.lex.o:(.bss+0x0): first defined here  CC      scripts/mod/empty.o
collect2: error: ld returned 1 exit status
make[3]: *** [scripts/Makefile.host:99: scripts/dtc/dtc] Error 1
make[2]: *** [/home/growflavor/ev3d10kern/ev3-kernel/scripts/Makefile.build:544: scripts/dtc] Error 2
make[2]: *** Waiting for unfinished jobs....
...
make[1]: *** [/home/growflavor/ev3d10kern/ev3-kernel/Makefile:1067: scripts] Error 2
make[1]: Leaving directory '/home/growflavor/ev3d10kern/ev3dev-buildscripts/build-area/linux-ev3dev-ev3-obj-4.19'
make: *** [Makefile:146: sub-make] Error 2
make: Leaving directory '/home/growflavor/ev3d10kern/ev3-kernel'

recursivetree, 2025-01-15:
I'm pretty sure I also encountered the `multiple definition of 'yylloc';` error on my journey of building newer versions of the kernel. If I remeber correctly, the issue stems from building an old kernel with a newer gcc version with stricter defualt settings. It has been patched in newer versions, so it is fine in 6.9, where I spent most of my attention. 

A quick google search pulled up this: https://github.com/torvalds/linux/commit/e33a814e772cdc36436c8c188d8c42d019fda639

Growflavor, 2025-01-16:
From the buildscripts README: 
"If you make changes to your local kernel configuration that you want to merge into the default configuration, run ./defconfig update. It will use the merge tool specified by the EV3DEV_MERGE_CMD environment variable."

@dlech
Regarding my interest to change the CONFIG_ZRAM_WRITEBACK 
when I make such changes via ./menuconfig & save ( which I presume  generates the local .config ? ) will running ./build-kernel use my modified .config or do I need to first run ./defconfig update before ./build-kernel ? or ? Thanks! ;-)

@recursivetree
Thank you...since I'm not sure what I can do about that patch you found...maybe I'll try using docker & an older debian with gcc prior to 10 to build rather than building directly from Ubuntu running on WSL2...

dlech, 2025-01-16:
> do I need to first run ./defconfig update before ./build-kernel ?

Build the kernel first. Then if it works as expected, run defconfig update to save the change. The defconfig change can then be committed with git.

Growflavor, 2025-01-16:
@ntxngle  re: "linux-image-4.19.85-ev3dev-3-beta2-ev3+_1_armel.deb"

Thank you!  I did install this kernel on one EV3 soon after you posted the link...
I will need to repartition my SDcard to create a partition for the /sys/block/zramX/backing_dev 

<details>
  <summary>if that works then hopefully I can assign it as a backing_dev & finally see if writeback behaves in a useful way...</summary>  
  
echo "to be created partition" > /sys/block/zram0/backing_dev

  ```
robot@ev3dev:~$ udevadm info -a -p $(udevadm info -q path -n /zram0 )

  looking at device '//devices/virtual/block/zram0':
    KERNEL=="zram0"
    SUBSYSTEM=="block"
    DRIVER==""
    ATTR{initstate}=="1"
    ATTR{backing_dev}=="none"
    ATTR{removable}=="0"
    ATTR{mm_stat}==" 6877184  2613810  3182592        0  3215360        5        1      104"
    ATTR{size}=="196608"
    ATTR{capability}=="10"
    ATTR{inflight}=="       0        0"
    ATTR{ro}=="0"
    ATTR{comp_algorithm}=="lzo [lz4] "
    ATTR{hidden}=="0"
    ATTR{io_stat}=="       0        0        0      731"
    ATTR{disksize}=="100663296"
    ATTR{alignment_offset}=="0"
    ATTR{discard_alignment}=="0"
    ATTR{stat}=="     902        0     7216     1400     2363        0    18904    11320        0     9030    12860        0        0        0        0"
    ATTR{range}=="1"
    ATTR{ext_range}=="1"
    ATTR{max_comp_streams}=="1"
  ```  
</details>

https://docs.kernel.org/admin-guide/blockdev/zram.html
https://www.reddit.com/r/linuxquestions/comments/qto7u3/repurposing_existing_swap_partition_as_zram/

ntxngle, 2025-01-17:
Quick warning to anyone who might try the kernel I posted: it breaks wifi. I wasn't able to undo it so I had to reinstall into SD card.

Growflavor, 2025-01-18:
> Quick warning to anyone who might try the kernel I posted: it breaks wifi. I wasn't able to undo it so I had to reinstall into SD card.

Thanks!  Also, to clarify, does the linux-image-4.19.85-ev3dev-3-beta2-ev3+_1_armel.deb contain the adjustments you made for controlling the motors at 12V or just the ZRAM writeback?

ntxngle, 2025-01-19:
Both @Growflavor