ev3dev/ev3dev #1662: zram-init how set ATTR{backing_dev} ?

Growflavor, 2025-01-18:
I'm stuck with how to set the zram "ATTR{backing_dev}=="

Hopeful, I created & mounted the partition: /dev/mmcblk0p6  1.0G   48K  1.0G   1% /swap

robot@ev3dev:~$` df -h
Filesystem      Size  Used Avail Use% Mounted on
udev             25M     0   25M   0% /dev
tmpfs            20M  352K   20M   2% /run
/dev/mmcblk0p2  7.4G  3.2G  3.9G  45% /
tmpfs            28M     0   28M   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs            28M     0   28M   0% /sys/fs/cgroup
/dev/mmcblk0p1   48M  230K   48M   1% /boot/flash
tmpfs           5.6M     0  5.6M   0% /run/user/1000
**/dev/mmcblk0p6  1.0G   48K  1.0G   1% /swap**

Then I attempted the following per https://www.kernel.org/doc/html/latest/admin-guide/blockdev/zram.html#writeback :

     robot@ev3dev:~$ echo /dev/mmcblk0p2 > sudo tee /sys/block/zram0/backing_dev

but still see:

     robot@ev3dev:/$ udevadm info -a -p /devices/virtual/block/zram0

  looking at device '//devices/virtual/block/zram0':
    KERNEL=="zram0"
    SUBSYSTEM=="block"
    DRIVER==""
    ATTR{initstate}=="1"
    ATTR{size}=="196608"
    ATTR{max_comp_streams}=="1"
    ATTR{hidden}=="0"
    ATTR{alignment_offset}=="0"
    ATTR{io_stat}=="       0        0        0     2447"
    ATTR{inflight}=="       0        0"
    ATTR{stat}=="    2514        0    20112     3720     9233        0    73864    29060        0    24850    32710        0        0        0        0"
    ATTR{range}=="1"
    ATTR{capability}=="10"
    ATTR{mm_stat}=="28225536 11093127 12103680        0 12324864      122      116      312"
    ATTR{ro}=="0"
    ATTR{comp_algorithm}=="lzo [lz4] "
    ATTR{ext_range}=="1"
    **ATTR{backing_dev}=="none"**
    ATTR{discard_alignment}=="0"
    ATTR{disksize}=="100663296"
    ATTR{removable}=="0"
_______________________________________________________________________________
So I also took a look at zram-init flags over at https://github.com/vaeth/zram-init/blob/main/sbin/zram-init.in

-b DEV     Use DEV as backing device
-I         If combined with "-b DEV", store incompressible pages to
           backing device
-w         If combined with "-b DEV", enable idle writeback to backing device

& made a guess at modifying the zram-init via

sudo nano /lib/systemd/system/zram_swap.service

robot@ev3dev:~$ cat /lib/systemd/system/zram_swap.service

`[Unit]
Description=swap with zram
DefaultDependencies=no
Before=swap.target

[Service]
Type=oneshot
RemainAfterExit=true

'# Change the maximal number of parallel streams (-s), algorithm (-a) and size (in megabytes) to your needs. Make sure to avoid collision of devices (-d...) with zram_tmp.service and zram_var_tmp.service and that num_devices in modprobe.d/zram.conf contains the maximal used number + 1

'#ExecStart=/usr/sbin/zram-init -alz4 -Lzram_swap 96
**ExecStart=/usr/sbin/zram-init -alz4 -b/dev/mmcblk0p6 -I -w -Lzram_swap 96**
ExecStop=/usr/sbin/zram-init 0

[Install]
WantedBy=swap.target`

robot@ev3dev:~$ 

robot@ev3dev:$ sudo systemctl daemon-reload
robot@ev3dev:$ sudo systemctl start zram_swap.service

but of course it fails to start with the flags I added since I could not figure out the proper syntax or if modifying the flags would even be one way to set the attribute...

I also could not figure out what makes a proper writeback partition (such as file system type, etc)...

Thus I'd be grateful for suggestions regarding what might be a proper way to actually go about creating a writeback partition & setting ZRAM0:

ATTR{backing_dev}=="none"

 to a partition such as

ATTR{backing_dev}=="/dev/mmcblk0p6"

Thanks!

dlech, 2025-01-18:
That feature was introduced in `zram-init` v7.0, ev3dev has v6.1.

Growflavor, 2025-01-19:
> That feature was introduced in `zram-init` v7.0, ev3dev has v6.1.

Thanks for the change log reminder ;-)  Since it is a script, I copied over zram-init v 7.1

I formatted /dev/mmcblk0p6 as ext4.

The writeback partition also needs to be unmounted.

robot@ev3dev:/$ sudo umount /dev/mmcblk0p6

The following then appears to work with the kernel that is built with the writeback enabled...

robot@ev3dev:/$ **sudo ./usr/sbin/zram-init -s2 -alz4 -b/dev/mmcblk0p6 -Lzram_swap 96**

_____________________________________________________
zram-init key changelog details:

https://github.com/vaeth/zram-init/blob/main/ChangeLog

*zram-init-7.1:
	**- Fix missing newline for backing_dev**

*zram-init-7.0:
	- Support discard also for swap
	- Support mem_limit
	**- Support backup_dev**

________________________________________________________
was useful to practice with making & manipulating linux partitions, kernel ./menuconfig & buildscript, process for understanding appropriate old script versions, & a bit more awareness about types of swap options...

closed 2025-01-19 by Growflavor