ev3dev/ev3dev #1664: 2025.01.22 Cannot Connect EV3 to Internet

ecschool, 2025-01-22:
2025.01.22 -- Internet Issue Report

Hello, Development Team. I would like some assistance in this matter. I already looked at the other posts/issues, and verified that my case is not an identical repeat.

I was trying to connect the EV3 to the internet via hardwire to my computer. But first, on the brick, I noticed something interesting:

[image]

So I hardwired it to my laptop (Windows 11 Home) via ethernet cable:

[image]

I opened an SSH terminal and updating it to install a package. I got this:

[image]

So I checked my internet connection settings:

[image]

I also decided to ping google.com to verify if I had connection (or not):

[image]

Not sure where to go from here. I want to install a package to test my webcam connected to the EV3 via USB for a much larger project, but I am stuck here.

Help would be greatly appreciated.

ecschool, 2025-01-22:
I forgot to add that I am using EV3-Sketch

dlech, 2025-01-22:
Did you set up internet connection sharing as described here: https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/?

ecschool, 2025-01-22:
Not yet. I was looking for this kind of document. Thank you, I will try it now.

Growflavor, 2025-01-24:
RE: "http://httpredir.debian" & EV3dev Stretch

I suggest that you update your /etc/apt/sources.list to the debian archive links...see this post:
https://github.com/ev3dev/ev3dev/issues/1608#issuecomment-1902726546

RE: your wifi photo & EV3 internet connections in general

EV3 has no onboard wifi chip thus requires something like a usb wifi adapter -- only a very few available ones work well 'plug & play' with both EV3dev & EV3-G...for example the one linked to at the following post which also has some internet connection tips (and EV3dev [maintains a list of other wifi adapters as well](https://github.com/ev3dev/ev3dev/wiki/USB-Wi-Fi-Dongles)):

https://github.com/orgs/ev3dev/discussions/1646#discussioncomment-10376087

some other posts with potentially helpful info -- like using hostname -I -- regarding troubleshooting 169. vs 192. internet connections:

https://github.com/ev3dev/ev3dev/issues/1618#issuecomment-1977099213