ev3dev/ev3dev #1665: Connect 3rd party i2c bmi160 gyro sensor

Muresan73, 2025-01-25:
I am trying to connect a bmi160 gyro sensor board with a modified rj12 cable. 

When I try to find the sensor address with `i2cdetect` the sensor does not show up.

I tried it with 65kohm and 91kohm pull up resistors ( 82kohm i didnt have at the moment ) but no luck.

When I tried to connect the sensor to a raspberry pi and esp32 both cases worked with and without pullup.
Then I tried to use the esp32 as a slave and that did show up on the ev3.

For this i am using the stable stretch image.

Do you guys had similar issues?
I dont have an oscilloscope but on standby with a multimeter seems to me the sensor is regulated to 3.3v

dlech, 2025-01-25:
What are you doing for power? The IO pins (pins 5 and 6) on the EV3 operate at 3.3V, but the power supply (pin 4 on the EV3) is 5V, which is more than the max rating of the the sensor.

I'm not sure how to troubleshoot the communication though without using a scope or logic analyzer. Maybe you could turn the ESP32 into one?

Muresan73, 2025-01-27:
I am using the 6DS3 board for the sensor, that has voltage regulator for 5V

I tried both esp32 and an arduino uno as logic analyzer but i am not getting values not sure what i am doing wrong, i will order one.

I wanted to clarify that 3.3v logic 1 should work with the ev3.

dlech, 2025-01-27:
> I am using the 6DS3 board for the sensor,

Adafruit 4503?

Muresan73, 2025-09-25:
yes, i am using specifically an STM LSM6DS3 but they are technically the same