ev3dev/ev3dev #1666: I can't update AND download any packages

stefantasticool, 2025-02-02:
**Describe the bug**
I can't download any packages
**To Reproduce**
Steps to reproduce the behavior:
I just typed sudo apt -get update and it didn't worked

```
Please paste a simple program to reproduce the problem here!
```

**Expected behavior**
I would expected that the ev3 will update. 

**Screenshots**
[image]
This shows up.

**Desktop (please complete the following information):**
 - OS: Windows10
 - Connection Type: USB
 - Other software: PuTTY

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: Latest

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

**Additional context**

dlech, 2025-02-02:
This is a known issue. Please see step 6.1 in the [getting started guide](https://www.ev3dev.org/docs/getting-started/).

stefantasticool, 2025-02-02:
It does the same thing. :(

dlech, 2025-02-02:
If it still says `httpredir` in the error message, then the apt `sources.list` file was not actually changed and you should try again.