ev3dev/ev3dev #1668: MS MUX with Pistorms

Noc1a, 2025-03-17:
**What are you trying to do?**

I am trying to use the mindsensors MUX with my Pistorms 2.

**What did you expect to happen**
The code posted bellow works well with an EV3 but doesn't seem to work with Pistorms.
Any hint of what I can do to make it work, please?

**What actually happened**

ev3dev2.DeviceNotFound: LegoPort(pistorms:BAS1:i2c80:mux1) is not connected.

**Source Code**

Here is the program I am trying to run.

```
    p =  LegoPort("pistorms:BAS1:i2c80:mux1")
    p.mode = "uart"
    p.set_device = "lego-ev3-ir"
    sleep(0.2)
    ir1 = InfraredSensor("pistorms:BAS1:i2c80:mux1")
    debug_print("prox:", ir1.proximity)
```

**Desktop (please complete the following information):**
Mac book pro

**Robot (please complete the following information):**
 - Device: Pistorms v2
```
Image file:         ev3dev-stretch-rpi2-generic-2020-04-10
Kernel version:     4.14.114-ev3dev-2.3.5-rpi2
Brickman:           0.10.3
BogoMIPS:           38.40
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=A020D4
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Plus Rev 1.4
BOARD_INFO_SERIAL_NUM=00000000c362a693
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=V2.09
BOARD_INFO_MODEL=PiStorms
BOARD_INFO_TYPE=aux
```

**Background Information**

dlech, 2025-03-17:
As mentioned in http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/pistorms.html#pistorms-in-port-mode-info:

> Sensors cannot be automatically detected, so you must manually specify the mode of the port before using a sensors. And in many cases, you must also manually specify the type of sensor that is attached.

So you will need to create a port object for `bas1_port = LegoPort("pistorms:BAS1")` (maybe there is a built-in constant for this?). Then with that port, `bas1_port.mode = "i2c-thru"` to put it in I2C mode to communicate with the MUX.

Then you need to do the equivalent of http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/i2c.html#manually-loading-devices

So something like: 

```python
# Not sure what the correct number is for the `0` in `i2c-0`.
with open("/sys/bus/i2c/devices/i2c-0/new_device", "w") as f:
    # Assuming sensor mux, not motor mux.
    f.write("ms-ev3-smux 0x50")
    f.write("ms-ev3-smux 0x51")
    f.write("ms-ev3-smux 0x52")

# Give the driver some time to load.
time.sleep(500)
```

Then you should be able to create a new `LegoPort()` object as you were trying to do. If the path still isn't correct, you should be able to look in `/sys/class/lego-ports/` to find out what the right one is.

Noc1a, 2025-03-17:
Thank you so much for your help, much appreciated!

I cannot write into new_device directly form python as it requires a root access: PermissionError: `[Errno 13] Permission denied: '/sys/bus/i2c/devices/i2c-1/new_device'`
Am I supposed to run the python app as root?

In the meantime, I did it in a terminal:
```
sudo bash
root@ev3dev:/home/robot# echo ms-ev3-smux 0x51 > /sys/bus/i2c/devices/i2c-1/new_device
```
but I got the following error message, normal?:
```
[  203.384296] nxt-i2c-sensor 1-0051: Could not read channel from mux (-121).
[  203.384364] nxt-i2c-sensor: probe of 1-0051 failed with error -121
[  203.384414] i2c i2c-1: new_device: Instantiated device ms-ev3-smux at 0x51
```

At the end, I still get the same error: 
'''
    bas1Port =  LegoPort("pistorms:BAS1", driver_name= "pistorms-in-port")
    sleep(1)
    bas1Port = "i2c-thru"
   
    # with open("/sys/bus/i2c/devices/i2c-1/new_device", "w") as f:
    #     # Assuming sensor mux, not motor mux.
    #     f.write("ms-ev3-smux 0x50")
    #     f.write("ms-ev3-smux 0x51")
    #     f.write("ms-ev3-smux 0x52")

    time.sleep(1)
    p =  LegoPort("pistorms:BAS1:i2c81:mux2")
'''

'ev3dev2.DeviceNotFound: LegoPort(pistorms:BAS1:i2c81:mux2) is not connected.'

Any idea what I can try next?

dlech, 2025-03-18:
> Am I supposed to run the python app as root?

No. Instead, you can change the permissions on the file. `sudo chgrp ev3dev /sys/bus/i2c/devices/i2c-1/new_device`. Then you should be able to write it from Python. You can also make a udev rule or other script so that this happens automatically at startup.

> bas1Port = "i2c-thru"

This line is wrong, it should be ` bas1Port.mode = "i2c-thru"`. So the port is probably in the wrong mode.

> Any idea what I can try next?

If still not luck communicating, you can use `i2cdump` to try poking around to see if you can actually communicate with the device. First, we will need to make sure the port is really in "i2c-thru" mode. Can do this via brickman on the screen or by reading/writing `/sys/class/lego-port/...`. This will let us make sure we actually got the I2C bus number and the address right.
 
Something like...

```
sudo i2cdump -r 0-7 0 0x50
```

Noc1a, 2025-03-19:
Thanks again, helped me a lot!

> bas1Port = "i2c-thru" oops sorry, my eyes got tired I guess :-)

I managed to make it work but it's still not very clean, let me explain you why.

First I created the Pistorms port
```
    bas1Port =  LegoPort("pistorms:BAS1", driver_name= "pistorms-in-port")
    bas1Port.mode = "i2c-thru"
```
Doing so, created 1 lego port and the 2 following sensors:
```
robot@ev3dev:~$ cat /sys/class/lego-sensor/sensor0/address
pistorms:BAS1:i2c80:mux1
robot@ev3dev:~$ cat /sys/class/lego-sensor/sensor1/address
pistorms:BAS1:i2c80
```
So no need to create 0x50 manually via new_device. 
Then I created 0x51 & 0x52 via new_device . This also created 2 sensors by address. However, the full address is as follow and not using the format `pistorms:BAS1:i2cXX`

```
bcm2835 I2C adapter:i2c81
bcm2835 I2C adapter:i2c81:mux
bcm2835 I2C adapter:i2c82
bcm2835 I2C adapter:i2c82:mux
```

Using this address on the sensor does work. But it's not very clean as it doesn't use the standard ev3dev format and that I cannot use the predefined built-in constants (e.g. INPUT1)

```
    p2 =  LegoPort("bcm2835 I2C adapter:i2c81:mux", driver_name="ms-ev3-smux")
    p2.mode = "uart"
    p2.set_device = "lego-ev3-ir"
    ir2 = InfraredSensor("bcm2835 I2C adapter:i2c81:mux")
```
Any idea how I could change it to the right format?

dlech, 2025-03-19:
> I managed to make it work

Very glad to hear that.

> Any idea how I could create the sensor with the right format?

If it is working, then it is the "right" format. There isn't a way to change it without modifying the Linux kernel drivers.

Instead, you could hide all of the not-so-clean parts in a separate Python module and make your own constants like `MUX_INPUT1 = "bcm2835 I2C adapter:i2c81:mux"`. Then at least your main program would look a bit cleaner.

Noc1a, 2025-03-20:
> If it is working, then it is the "right" format. There isn't a way to change it without modifying the Linux kernel drivers.
Hehehe true! OK understood. Will do as suggested, thx a lot! I now understand better how the devices are handled (I am new with ev3dev)

Also, for information, I had to open and close the special file new_device for each device creation (didn't work to write 3 lines at once). 
Here is the full procedure in case anyone else tries to do it:

**Make new_device writable:**
Create a script e.g.  /home/robot/startup/new_device_rights.sh with:
```
#!/bin/bash
sudo chgrp ev3dev /sys/bus/i2c/devices/i2c-1/new_device
sudo chmod g+w /sys/bus/i2c/devices/i2c-1/new_device
```
Run the script at startup (I chose here cron):
`sudo crontab -e
`
and add the line:
`@reboot /home/robot/startup/new_device_rights.sh
`
**And here is the raw code:**

  ```
    device_path = "/sys/bus/i2c/devices/{:01d}-{:04x}"

    if not os.path.exists(device_path.format(1, 0x50)):
        bas1Port =  LegoPort("pistorms:BAS1", driver_name= "pistorms-in-port")
        bas1Port.mode = "i2c-thru"
        sleep(0.2)

    if not os.path.exists(device_path.format(1, 0x51)):
        try:
            with open("/sys/bus/i2c/devices/i2c-1/new_device", "w") as f:
                f.write('ms-ev3-smux 0x51\n')
        except Exception as e: 
            debug_print( e)
 
    if not os.path.exists(device_path.format(1, 0x52)):
        try:
            with open("/sys/bus/i2c/devices/i2c-1/new_device", "w") as f:
                f.write('ms-ev3-smux 0x52\n')
        except Exception as e: 
            debug_print( e)

    sleep(0.2)

    p =  LegoPort("pistorms:BAS1:i2c80:mux1", driver_name="ms-ev3-smux")
    p.mode = "uart"
    p.set_device = "lego-ev3-ir"
    sleep(0.2)
    ir1 = InfraredSensor("pistorms:BAS1:i2c80:mux1")
    debug_print("prox:", ir1.proximity)
    
    p2 =  LegoPort("bcm2835 I2C adapter:i2c81:mux", driver_name="ms-ev3-smux")
    p2.mode = "uart"
    p2.set_device = "lego-ev3-ir"
    sleep(0.2)
    ir2 = InfraredSensor("bcm2835 I2C adapter:i2c81:mux")
    debug_print("prox:", ir2.proximity )

    p3 =  LegoPort("bcm2835 I2C adapter:i2c82:mux")
    p3.mode = "analog"
    p3.set_device = "lego-ev3-touch"
    sleep(0.2)
    ts = TouchSensor("bcm2835 I2C adapter:i2c82:mux")
    debug_print("is pressed:", ts.is_pressed & 0XF )

```

I will write a proper driver for this SMUX, for Pistorms and EV3 as the values of the sensors need to be adapted. Any interest of committing it in an ev3dev git repo or what is the usual way to share this kind of code ?

Growflavor, 2025-03-21:
> "I will write a proper driver for this SMUX, for Pistorms and EV3 as the values of the sensors need to be adapted."

If you have also used the OEM EV3 brick, I'm very interested to hear your real life experience with benefits/cons using EV3dev with a Pistorms2 with the Pi 3B+ vs EV3dev on an OEM EV3 brick.  Thanks!

& Please share what you come up with as at least a link to your github repo...I personally have 'overlooked' Pistorms so far mainly due to what you are clarifying here...so depending on your experience with the Pistorms 2 & Pi 3B+ maybe I might need to reconsider discussing this option with my son...especially now that we have your example code to work with...

Noc1a, 2025-03-29:
Sure, so I got an EV3 brick overclocked @ 456MHZ and Pistorms with Pi 3B+ 1.4GHZ quad core.

**Performance**
The difference is **HUGE**. To give you some concrete examples, I just timed both devices:

Boot time:
```
EV3: 80s
Pistorms: 9s
```

Execute a random python application:
```
EV3: 14s
Pistorms: 3s
```
Performance is the main reason I bought it. With its quad core it can handle very well multiple threads reading different sensors (odometry, distance, ...). Also I guess it should have enough power to do SLAM with a LIDAR etc (one of my next project).

**Color Touch Screen**
Quite nice as you can create your own user interface. Especially for a kid as you can create games etc.

**GPIO**
_You can use it jointly with other hardware that uses GPIO pins, provided there is no conflict between Raspbery Pi pins used by PiStorms_
This mean that you can go further by using non lego devices.

**Set-up**
I did struggle a bit to make everything work, but as one said, once we know it's very easy :-). As you show some interests, I will try to do a tutorial on the main site to share some basic information on how to get up to speed with Pistorms.

closed 2025-03-29 by Noc1a

Growflavor, 2025-03-29:
RE: "Performance The difference is HUGE."

RE: "As you show some interests, I will try to do a tutorial on the main site to share some basic information on how to get up to speed with Pistorms."

Fantastic!  I'll look forward to it...if possible please post a link here in this thread (makes it easier for me to not miss it in my email).  Thanks!