ev3dev/ev3dev #1671: Controling EV3 robot with a keyboard connected to a PC

bettettet, 2025-05-24:
**Question**

How do you control a robot wih a keyboard? (make it so that it moves when you press wasd)

**Background Information**

I have seen youtube videos where people control a robot using a keyboard connected to a PC, but Im not able to get it to work.
[Example](https://www.youtube.com/watch?v=vhmM5O51TuQ)
I want to make a python program, which says which keys are being pressed on the keyboard using the EV3 speaker

**Screenshots**
[image]
Right now I have a program, which _should_ print keypresses and also say them out loud with the EV3 speaker, but its not printing anything and not making any sounds
Output: (the "wasdwasd" is me pressing keys, and it should print the keys below but it doesntprint anything)
[image]

**Desktop:**
 - OS: Windows 10
 - Connection Type: Bluetooth
 - Software: PuTTy

**Robot:**
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          4.0
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016536420DE
BOARD_INFO_TYPE=main

JSDaleman, 2025-05-24:
The teleoperation you want to perform requires connecting the EV3 to either a server or your PC. To do this, you can create the interconnection through a server or by sending the data from the PC to the robot using a Python script. If you run it that way, the EV3 will look for the available hardware directly—its keyboard, as you saw in the tests you performed.

For example, I did it using ROS and a mqtt server: https://github.com/JSDaleman/ev3_mqtt_ros. Sorry that the documentation is in Spanish—if you need help, try it like I did, and don't hesitate to ask me.

bettettet, 2025-05-24:
Can you show me how to do it with the python script, sending the data from the PC to the robot?

JSDaleman, 2025-05-24:
For example with MQTT, I use this videos
https://youtu.be/nTJBxofQLxk?si=zJWH89hMo4poSNG1

You have other options like use a serial comunication with the Bluetooth or with a cable.

dlech, 2025-05-24:
The problem is that you are using a remote terminal, which doesn't know anything about the keyboard on your computer.

So to make that work with the `keyboard` module, you would have to unplug the keyboard from your computer and plug it into the USB port on the EV3! But I have a feeling that is not what you actually want to do.

Instead, you need to use a module that can read terminal input instead of keyboard input. For this, I would suggest to use the `curses` module from the Python standard library. https://docs.python.org/3.5/howto/curses.html

Also, you should always run your scripts with `brickrun`. If you want to have your program on the EV3 interact with the remote terminal, use `brickrun -r`, otherwise it will use a virtual terminal on the EV3. It will do thing like stop motors if your program crashes. 

And if you should never need to run your own programs as root. If you do, there is probably a better way.