ev3dev/ev3dev #1676: I'm not able to connect to my EV3 with EV3dev on it.

Gemsa-developer, 2025-07-30:
**What are you trying to do?**

I am trying to connect to my ev3 using SSH. I'm using bluetooth tethernig, and as last resort i've tried usb tethering

**What did you expect to happen**

It should have connected to my brick.

**What actually happened**

Terminal showed message "ssh: connect to host ev3dev.localhost port 22: Connection refused" or "ssh: connect to host ev3dev.localhost port 22: Connection refused"

**Desktop (please complete the following information):**
 - OS and version: MX linux (debian based distro) verion 23.6 x86
 - Connection Type: USB tethering or Bluetooth tetering
 - Software: Build-in terminal

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: 2020 verion (newst avalible on webpage)

**Background Information**

Yesterday I've booted my brick with ev3dev for the first time, and from there I am trying to connect to my brick, from pc which is used for tethering too, using SSH. When i type ssh command, I'm greeted with an error (look to an screenshot and "what happend page). i've tried every name combination, i could think of (ip only, robot@ev3dev, robot@ev3dev.local, robot@ev3dev.localhost, robot@IP, ev3dev, ev3dev.local). i didn't change anything in ev3 img. ive tried connecting from other pc with no luck either. 

Thank You all in advance
ANTon

Screenshots:
[image]
[image]

dlech, 2025-07-30:
> I'm using bluetooth tethernig, and as last resort i've tried usb tethering

Tethering is only useful for daisy-chaining EV3s, not for connecting to it from a computer. So turn tethering off and then follow the instructions for sharing your computer's internet connection to the EV3 and you should have better luck.

If you did it correctly, the EV3 should get an IP address that does _not_ start with `169.x.x.x` (the actual starting number depends on the OS you are connecting to). If connecting by hostname doesn't work, you can try connecting to that address. If you continue to get an IP address that starts with 169., that means that the EV3 got a self-assigned IP address because the computer was not set up correctly for Internet connection sharing.

Gemsa-developer, 2025-08-01:
So I've got usb WiFi card, and it works now.